z88dk/z88dk #412: CYGWIN Makefile and Z88DK

Fabrizio-Caruso, 2017-09-26:
I am trying to use CYGWIN Makefile to compile with Z88DK but I am getting erros about not found files
when running a the comand line:
/cygdrive/c/z88dk/bin/zcc.exe +ace -O3 -D__ACE__ -clib=ansi  -Cz--audio -o min -create-app  $(SOURCE_PATH)/display_macros.c $(SOURCE_PATH)/enemy.c $(SOURCE_PATH)/invincible_enemy.c $(SOURCE_PATH)/level.c $(SOURCE_PATH)/character.c $(SOURCE_PATH)/text.c $(SOURCE_PATH)/missile.c $(SOURCE_PATH)/strategy.c $(SOURCE_PATH)/input_macros.c $(SOURCE_PATH)/main.c

ERRORS:

PROCESSING ./src/display_macros.c
zcpp -iquote"." -DZ80 -DACE -D__ACE__ -D__ACE__  -DSCCZ80 -DSMALL_C -D__SCCZ80 -isystem"c:/z88dk//include"    "./src/display_macros.c" "C:\cygwin64\tmp\zccA5822.i2"
make: *** [Makefile:78: ace_11k] Error 1

suborb, 2017-09-27:
Is this a cygwin compiled zcc or a mingw compiled zcc?

I'm just wondering if the different behaviour in paths i.e. /cygdrive vs c: is causing problems. I know @zx70 uses mingw fairly successfully.

Fabrizio-Caruso, 2017-09-28:
[PROBLEM SOLVED] I have managed to solve the problem by reorganizing my code.
I now void implementations in .h and use relative paths (ugly) to include some headers.

closed 2017-09-28 by Fabrizio-Caruso