z88dk/z88dk #415: Chained m4 defines - fail

feilipu, 2017-09-27:
Defines created in `config_*.m4` chain properly for one reference, but don't chain for the second reference. This also causes the rest of the second file to be incorrectly read.

The [`config.m4`](https://github.com/feilipu/z88dk/blob/75b861044023915bb853011628096074ce00d3d0/libsrc/_DEVELOPMENT/target/yaz180/config.m4#L39) includes a chain of definition in `config_target.m4` to `config_82c55.m4` to `config_ide.m4`.

Note: `config_ide.m4` has been excluded from #414 request to stop it being broken.

```m4
include(`target/z88dk_version.m4')
include(`target/yaz180/config/config_target.m4')
include(`target/yaz180/config/config_cpu.m4')
include(`target/yaz180/config/config_clib.m4')
include(`target/yaz180/config/config_clib_const.m4')

include(`target/yaz180/config/config_82c55.m4')
include(`target/yaz180/config/config_am9511a.m4')
include(`target/yaz180/config/config_pca9665.m4')

dnl# include(`target/yaz180/config/config_ide.m4')
```
definitions in [`config_target.m4`](https://github.com/feilipu/z88dk/blob/75b861044023915bb853011628096074ce00d3d0/libsrc/_DEVELOPMENT/target/yaz180/config/config_target.m4#L42)
```m4
define(`__IO_PIO_PORT_BASE', 0x4000)        # Base Address for 82C55
```
 can be used in [`config_82c55.m4`](https://github.com/feilipu/z88dk/blob/75b861044023915bb853011628096074ce00d3d0/libsrc/_DEVELOPMENT/target/yaz180/config/config_82c55.m4#L7).
```m4
define(`__IO_PIO_PORT_A',   `__IO_PIO_PORT_BASE+0x00')  # Address for Port A
```
But, adding a second level of abstraction via [`config_ide.m4`](https://github.com/feilipu/z88dk/blob/75b861044023915bb853011628096074ce00d3d0/libsrc/_DEVELOPMENT/target/yaz180/config/config_ide.m4#L14) doesn't correctly associate the defines found in this file. It leaves them assigned to themselves. Even those with simple numeric assignments are not converted properly
```m4
define(`__IO_PIO_IDE_LSB',   `__IO_PIO_PORT_A')   # IDE lower 8 bits
```
Also, perhaps related, trying to compile a program that doesn't use the defines at all fails and creates a core dump, pointing to [`modlink.c` line 395](https://github.com/z88dk/z88dk/blob/master/src/z80asm/modlink.c#L395).

```bash
z80asm --cpu=z180 -b -d  -o"time_app" -m -s -L.  -L"/home/phillip/Z80/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/lib/sdcc_iy"   -D__SDCC -D__SDCC_IY -I"/home/phillip/Z80/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/target/yaz180" --list  -ilib/yaz180/time -iyaz180    "/tmp/tmpXXwyy0Xw.o" "/tmp/tmpXXRjjEFx.o"
z80asm: modlink.c:395: patch_exprs: Assertion `expr->target_name == NULL' failed.
Aborted (core dumped)
Errors in source file /home/phillip/Z80/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/target/yaz180/yaz180_crt.asm:
phillip@SilverQuad:~/Z80/src/atest$ zcc +yaz180 -subtype=basic_dcio -v --list -m -SO3 --c-code-in-asm --opt-code-size -clib=sdcc_iy -llib/yaz180/time --max-allocs-per-node200000 @atest.lst -o time_app -create-app
```

feilipu, 2017-09-27:
Have tried to fix this using `eval()`, but it doesn't help.
```m4
define(`__IO_PIO_PORT_BASE', 0x4000)        # Base Address for 82C55
```
```m4
define(`__IO_PIO_PORT_A',   eval(__IO_PIO_PORT_BASE+0x00))  # Address for Port A
```
```m4
define(`__IO_PIO_IDE_LSB',   eval(__IO_PIO_PORT_A))   # IDE lower 8 bits
```
But, even using `eval()` at each stage, adding a second level of abstraction via `config_ide.m4` doesn't correctly associate the defines found in this file. Even those defines with simple numeric assignments in `config_ide.m4` are not converted properly. This is the result, in each case.
```asm
defc __IO_IO_PIO_IDE_LSB = __IO_IO_PIO_IDE_LSB
```

aralbrec, 2017-09-27:
These should be done without the quoting:
```
define(`__IO_PIO_IDE_LSB',   `__IO_PIO_PORT_A')   # IDE lower 8 bits
```

So:
```
define(`__IO_PIO_IDE_LSB',   __IO_PIO_PORT_A)   # IDE lower 8 bits
```

If you quote `__IO_PIO_PORT_A` the string gets stored in the macro and is evaluated later which may cause trouble with some of the syntax.  Without the quoting, `__IO_PIO_PORT_A` is evaluated to a constant right in the definition.

For similar reason this should be done without quotes for the value part:
```
define(`__IO_PIO_PORT_A',   `__IO_PIO_PORT_BASE+0x00')  # Address for Port A
```

An eval will compute the value and store that rather than storing a string `a+b` that may cause trouble later.

aralbrec, 2017-09-27:
I've committed the necessary changes to the yaz180 config files so please check them out.

The m4 eval function can generate output in any radix so in many places I've evaluated the constants in hex and for that you have to add `0x` to the front of the result.  You will see a back tick/forward tick between the `0x` and `eval` which is just a way to separate the two strings so that m4 can see them as two instead of one long word.

closed 2017-09-27 by aralbrec

feilipu, 2017-09-28:
Thanks. Works perfectly.
Fixed [a few items](https://github.com/z88dk/z88dk/pull/420/commits/561838c9329fb833f777e1f966530c222f3a573f), and updated rc2014 to use the same system in #420.

commit 8e767120 (2017-09-27): Fix #415 +yaz180: config files repaired