z88dk/z88dk #418: sccz80:  incorrectly extending unsigned char into int

aralbrec, 2017-09-28:
Example:

```
extern int a;
extern unsigned char compute(void);

void main(void)
{
	a = compute();
}
```

zcc +zx -vn -a -clib=new test.c

```
._main
	call	_compute
	ld	(_a),hl
	ret
```

The result of `compute` is in register `l` yet the entire `hl` is saved to variable `a` without setting `h` to zero first.

suborb, 2017-09-28:
I think this is correct for functions generated by sccz80. In that case hl is fully formed and sign extended as appropriate. I believe classic library written in assembler does the same thing.

However I'm guessing that sdcc doesn't do that? And the parts of newlib that are being borrowed also don't.

aralbrec, 2017-09-28:
Hmm yeah.  The new library functions will often be returning in L only.

The same code is generated if the return type for `compute` is a signed char rather than an unsigned one.  I'm not so sure it's the case that code returning signed chars are sign extending to int before return from the classic library asm functions.

suborb, 2017-09-28:
Yeah, sccz80 basically has a 16 bit return or a 32 bit return value. It's a hangover from the origins and K&R, but has the side effect that it means we don't need to liberally scatter compiled code with `call l_sxt` or `ld h,0`

Actually, I don't think there's that many at all:

`find . -name '*.h' | xargs grep "extern[ \t]*char" | grep -v "_[ \t]*\*"`

gives strpkbrk and strpbrk_callee which means the prototypes are wrong.

The rest of them can be shown up by switch char for `unsigned char` or `uint8_t` and doing a bit of eyeball filtering.

What's left is mostly arch/ stuff and screen manipulation. I'd say about 50 or so are affected and they are unsigned char.

So, we can blow 7T on an indiscriminate ld h,0 or only do it for sccz80 and remove the possibility of mixing object files.

suborb, 2017-11-01:
I've added handling for `__z88dk_sdccdecl` so that sccz80 extends l out into the full word for function calls that return a char.

Thus, although we have different calling conventions, sccz80 should be able to successfully call code compiled with sccz80 and sdcc.

closed 2017-11-01 by suborb