z88dk/z88dk #421: cpcrslib not recognized as a lib by Z88DK

Fabrizio-Caruso, 2017-09-28:
I have downloaded the latest version of cpcrslib for Z88DK (not the one for sdcc) and I am trying to figure out how to use it.

It seems that Z88DK is not recognizing cpcrslib as a correct lib file:
"
Error: file './cpcrslib.lib' not a library file
1 errors occurred during assembly
"

Has anyone  managed to use cpcrslib with a recent version of Z88DK?

pauloscustodio, 2017-09-28:
Can you run ```z80nm cpcrslib.lib```? What is the library version?
z80asm issues that error when the library is a non-supported version.

suborb, 2017-09-28:
Z80asm has changed a lot this year including the library file format so the library will need recompiling to work with master.

Fabrizio-Caruso, 2017-09-28:
@pauloscustodio I use the latest cpcrslib for Z88DK with a very recent version of Z88DK.
It would be nice to have a compatible version of cpcrslib.

suborb, 2017-09-28:
Have you tried this? Download cpcrslib_z88dk.zip from sourceforge and then adjust the source for the scope directive changes as follows:

```
% unzip cpcrslib_z88dk.zip
% cd cpcrslib
% gsed -i s,XLIB,GLOBAL,g *
% gsed -i s,LIB,GLOBAL,g *
% gsed -i s,XREF,GLOBAL,g *
% gsed -i s,XDEF,GLOBAL,g *
% z80asm -v -xcpcrslib.lib @cpcrslib.lst
```

pauloscustodio, 2017-09-28:
Maybe I should put pack the deprecated XLIB, LIB, XREF and DEF?

What do you think?

Fabrizio-Caruso, 2017-09-28:
Is there are technical reason why cpcrslib is not simply part of Z88DK? Is the license incompatible?

Fabrizio-Caruso, 2017-09-28:
@suborb, gsed is not available in Cygwin installation. Where can I get it? Any alternatives?

pauloscustodio, 2017-09-28:
I will add back the deprecated XLIB, LIB, XREF and DEF.

I tend to have a z88dk view of the world and forget about the users outside. Sorry.

suborb, 2017-09-28:
@Fabrizio-Caruso try `sed` instead - I'm on a Mac so have to use GNU sed (gsed) for the -i option rather than BSD sed.

pauloscustodio, 2017-09-28:
I've committed a new version of z80asm that puts back the deprecated XDEF, XLIB, XREF and XLIB, and managed to build the cpcrslib.lib:   
```
>z80asm -v -xcpcrslib.lib @cpcrslib.lst
....
Creating library 'cpcrslib.lib'
```

closed 2017-09-28 by pauloscustodio