z88dk/z88dk #423: Error at file 'fseek.c' line 60: symbol '_fseek1' not defined

electrified, 2017-09-28:
This program:

```
#include <stdio.h>

int main(int argc, char **argv)
{
  FILE *fpsave;
  fpsave = fopen(argv[1],"rb");

  fseek(fpsave, 0, SEEK_END);
  return 0;
}
```

Compiled with:
`zcc +cpm test.c -o test -create-app`

Results in:

`Error at file 'fseek.c' line 60: symbol '_fseek1' not defined`

This was working previously (with a git pull from early August), I think this commit might have broken it: https://github.com/z88dk/z88dk/commit/60b348f092199b1ece7d3c5919984c16e14e5e9b but I'm not sure what's wrong exactly.

Thanks,
Ed

aralbrec, 2017-09-29:
Yes it looks like the asm label "fseek1" is missing a leading underscore.  There may also be a section assignment problem since the c compiler may close the code section after the c code is generated.

I'll leave it to @suborb to have a look.  Is the c code in the classic library only compiled by sccz80?

suborb, 2017-09-29:
Correct, the classic C code is only compiled by sccz80. As a result the section is still open.

closed 2017-09-29 by suborb