z88dk/z88dk #424: Library equivalents for multi-byte load/store - Question

feilipu, 2017-09-29:
Working on ChaN's FatFs code, there is a lot of usage of simple load/store routines (written to be endian safe), but as Z80 is a LE CPU, probably there are easier ways to do below within the z88dk internal library.

Is there a preferred solution?

```asm
/*-----------------------------------------------------------------------*/
/* Load/Store multi-byte word in the FAT structure                       */
/*-----------------------------------------------------------------------*/

static
WORD ld_word (const BYTE* ptr)    /*     Load a 2-byte little-endian word */
{
    WORD rv;

    rv = ptr[1];
    rv = rv << 8 | ptr[0];
    return rv;
}

static
DWORD ld_dword (const BYTE* ptr)    /* Load a 4-byte little-endian word */
{
    DWORD rv;

    rv = ptr[3];
    rv = rv << 8 | ptr[2];
    rv = rv << 8 | ptr[1];
    rv = rv << 8 | ptr[0];
    return rv;
}

static
QWORD ld_qword (const BYTE* ptr)    /* Load an 8-byte little-endian word */
{
    QWORD rv;

    rv = ptr[7];
    rv = rv << 8 | ptr[6];
    rv = rv << 8 | ptr[5];
    rv = rv << 8 | ptr[4];
    rv = rv << 8 | ptr[3];
    rv = rv << 8 | ptr[2];
    rv = rv << 8 | ptr[1];
    rv = rv << 8 | ptr[0];
    return rv;
}

static
void st_word (BYTE* ptr, WORD val)    /* Store a 2-byte word in little-endian */
{
    *ptr++ = (BYTE)val; val >>= 8;
    *ptr++ = (BYTE)val;
}

static
void st_dword (BYTE* ptr, DWORD val)    /* Store a 4-byte word in little-endian */
{
    *ptr++ = (BYTE)val; val >>= 8;
    *ptr++ = (BYTE)val; val >>= 8;
    *ptr++ = (BYTE)val; val >>= 8;
    *ptr++ = (BYTE)val;
}
static
void st_qword (BYTE* ptr, QWORD val)    /* Store an 8-byte word in little-endian */
{
    *ptr++ = (BYTE)val; val >>= 8;
    *ptr++ = (BYTE)val; val >>= 8;
    *ptr++ = (BYTE)val; val >>= 8;
    *ptr++ = (BYTE)val; val >>= 8;
    *ptr++ = (BYTE)val; val >>= 8;
    *ptr++ = (BYTE)val; val >>= 8;
    *ptr++ = (BYTE)val; val >>= 8;
    *ptr++ = (BYTE)val;
}
```

aralbrec, 2017-09-29:
You could try replacing those with macros:

```
#define ld_word(a)  (*(WORD*)(a))
#define ld_dword(a)  (*(DWORD*)(a))
#define ld_qword(a)  (*(QWORD*)(a))

#define st_word(p,v)  (*(WORD*)(p) = (WORD)v)
#define st_dword(p,v)  (*(DWORD*)(p) = (DWORD)v)
#define st_qword(p,v)  (*(QWORD*)(p) = (QWORD)v)
```

Program size may increase for the QWORD case because we don't have as many code size reduction rules for 64-bit numbers.  It may be an advantage to keep the qword functions and replace the code with assignments.   The same may be said for DWORD but the situation is better for that (and I have a bunch more rules for 32-bit coming).  sccz80 will have to use the qword functions as it doesn't have a 64-bit type.

feilipu, 2017-09-29:
OK. Thanks.

I'll add these macros in alongside the [`string.h` macros](https://github.com/feilipu/z88dk-libraries/blob/master/ff/source/ff.c#L40), I already did, then convert the FatFs functions to macros.

Should make a substantial speedup (I hope).

closed 2017-09-29 by feilipu