z88dk/z88dk #425: appmake: banked memory issues

aralbrec, 2017-09-29:
Some things that have come up:

1. Sections "BANK_00" through "BANK_07" (numbers with leading zeroes) cannot be used anymore.

The bank numbers used to be two-digit hex and now they are decimal as in "BANK_0" through "BANK_7". I thought compatibility was kept because the section names are scanned as "BANK_%d" to identify number but the incompatibility happens at link time where the section names have no meaning and are treated as strings.  So BANK_00 is placed in a separate binary outside the memory map because it is not the same as BANK_0.

Section names without the leading zeroes must be used from now on.

2. Stuff placed in sections not part of the memory map and without org gets appended to the last defined section.

From the first point, someone was using a section "BANK_05" which was not being placed in BANK_5 but rather in the last defined section in the memory map "DIV_31".  This was a mysterious error and there is no easy way to find out what was placed in DIV_31 to know where the data was coming from.  The map file showed nothing assigned to section "DIV_31" (that data was in BANK_05) so you couldn't find out what was in there.

**Is it possible to add to the map file information on what the parent section is**, as in the section holding the org?  This way the map file could have shown that the data was in BANK_05 but its parent was DIV_31 so it would be known BANK_05 would be part of the DIV_31 binary.

I'm also adding section "UNASSIGNED" to the end of the memory map for all targets so these things are made part of an invalid binary rather than part of a legit binary in the wrong place.  Appmake's banking code will remove section UNASSIGNED from the memory model and emit a warning.

3. It used to be that a user defined section with its own org was output as a separate binary.

This is still the case.  But appmake's banked model treats sections not part of a bank space as belonging to the main bank.  So that means any processing by appmake involving banked memory will put that user defined section into the main bank too.

If that's not what is wanted, the section should be removed from appmake's memory model by adding `--exclude-sections "space_separated_section_names_here"` to the appmake invoke.  If invoking from zcc, this would be added with `-Cz'--exclude-sections "..."'`.   By excluding particular sections, those binaries will be ignored by appmake and will be left in the user directory.

4. There's a new requirement - being able to assemble to a specific address but then being able to place the resulting binary someplace else.

The specific use-case is someone assembling for org 0x2000 but wanting to store the binary someplace else so that it can be memcopied to the right address on demand.

The construction of output binaries and knowledge of memory maps is in appmake so this is a requirement for appmake.

I have one idea to solve this now :- z80asm continues to generate binaries using the org given but the physical address for storage is placed in the section name.  So a section name like "UTILITY_PORG_0x4567" contains a physical org statement.  Appmake would scan the 0x4567 and change its org address to that in its memory model so that it places that section at the physical address.  This would work in different bank spaces too (the bank space is also a substring of the section name).

However, this makes the porg part of the section name which means the section name would have to be changed if the porg is changed and references to the section name may be spread all over the code base in different files.  So it would be better if this porg were in one place as part of the memory map, beside the regular org.

Another idea is this:  while reading the memory map, z80asm changes the name of the section to include the porg in the section name.

```
SECTION UTILITY
org 0x2000
porg 0x4567
```

z80asm would read this as:

```
SECTION UTILITY_PORG_0x4567
org 0x2000
```

I'm not sure if there is a better way to handle this.

suborb, 2017-09-29:
Just to confirm, this also affects +sms?

aralbrec, 2017-09-29:
It affects +zx, +zxn and +glue (not attached to a specific target) now.   I haven't modified +sms yet but it will eventually affect +sms too.

aralbrec, 2017-10-01:
#429 Adds keywords PHASE and DEPHASE to alter the current PC value for assembling while keeping the physical org the same.

Is the PHASE value a property of the section?  So if in a single file I do:

```
SECTION PART_1
PHASE 0x8000

ld hl,10
ld de,100
ld bc,256

SECTION PART_2

defm "blah blah blah",0

SECTION PART_1
ldir
```

Does the secion PART_1 part with the ldir in it still have PHASE applied?  Or was PHASE closed by the intervening section?  Does section PART_2 not get PHASE applied because PHASE applies only to PART_1?

aralbrec, 2017-10-01:
When the user is responsible for setting an absolute address for PHASE, he's effectively doing layout in memory by hand because if code is spread across several files, he must calculate a PHASE value for each bit so that the independent pieces don't overlap at the PHASE address.

Is it possible to attach a default phase address to a section?  The way things work now sections have an implied phase of 0 and then the user can change that with PHASE directive and it goes back to 0 for DEPHASE.  If there were a default phase address for a section, then you could just add to that specific section from many different files and everything would be packed together correctly for the PHASE address.

In particular I'd want several c files to be compiled to a specific section that would be phased so that the code is stored at a physical address determined by the section org but the addresses would be set by the section phase so that they could be moved at runtime to the correct address for execution.  Then there's no hand layout for things spread across files.

pauloscustodio, 2017-10-01:
> Is the PHASE value a property of the section? 

Yes. The PHASE/DEPHASE blocks are a property of the section and the current phase state is remembered during a compilation unit, even if other sections are written in between. 

What PHASE does is create a new absolute ASMPC at the start address, and all the labels inside a phase block are defined as DEFC instead of relocatable addresses.

> Does the section PART_1 part with the ldir in it still have PHASE applied? 

Yes, the section PART_1 still has the phase in place.

pauloscustodio, 2017-10-01:
> In particular I'd want several c files to be compiled to a specific section that would be phased so that the code is stored at a physical address determined by the section org but the addresses [...]

I do not understand this. Could you provide a use case?

I understand the need to store different bits of C code to a specific memory bank, so a section physically located at 0x2000 would receive code from multiple C source files and pack them together. In the output there will be a bank_nn.bin file, that the loader has to load to the correct memory bank py paging memory and loading the bin file. I don't understand why the memcopy is needed.

aralbrec, 2017-10-01:
We're still talking about the same need - wanting to pack code into a another memory bank for transfer later to a different address.

With C and asm you can place it in a different bank so that physical address == org address.  This works now.

But now people want to org at 0x2000 (to use a concrete number) but store it in a different memory bank, say BANK 6 with org 0xc000, mixed together with other stuff meant for BANK 6.  Then they can copy it later to 0x2000 as needed.  I don't question the why - this is just something people want to do.  When the binary is built, that stuff will be in BANK 6 which makes it convenient to load / use.

So PHASE/DEPHASE is solving this problem within one file.

But suppose my source is split between two files or between c and asm and I want to compile all of this with PHASE 0x2000 destined for BANK 6?

FILE1
```
SECTION BANK_6
PHASE 0x2000

part1code:
   ld hl,10
   ret
```

FILE2
```
SECTION BANK_6
PHASE 0x2000+??????

part2code:
   ld bc,10
   ret
```

I actually want these two parts to be placed next to each other with PHASE 0x2000, ie I effectively want this:

```
SECTION BANK_6
PHASE 0x2000

part1code:
   ld hl,10
   ret
part2code:
   ld bc,10
   ret
```

Because the code is split into two files I have to place part2 by hand.  part1 will be PHASE 0x2000 to place at address 0x2000 and part2 will be PHASE 0x2000 plus the size of part1.  This is what I mean by the user having to do hand placement of the PHASE org to get things in different files to pack together.

So as another option, the idea is to give the entire section a PHASE so that the linker can put all the pieces of the section together and apply a PHASE to all of it.  Then there is no need for hand layout across files.

The memory map would look like this:

```
SECTION BANK_6
org 0xc000
SECTION BANK_6_STORED
PHASE 0x2000
```

Because section `BANK_6_STORED` does not have its own org, it is output as part of the `BANK_6` binary but everything inside that section has PHASE 0x2000 applied.  Then I can assign to this phased section from multiple files and the linker can pack it together.

I am still wondering if this is the best way to do things.  For example, another way to do this is just make a new section:

```
SECTION BANK_6_STORED
org 0x2000
```

And assign stuff there.  Then the linker will output a separate `BANK_6_STORED` binary.  The user can then physically add this to the `BANK_6` binary with `cat BANK_6_STORED.bin >> BANK_6.bin`.  Then delete the `BANK_6_STORED.bin` so that it is not found by the next step.  The next step is a call to `appmake` to form the final output.  This way `BANK_6_STORED` will become part of `BANK_6`.

But to get the address of the appended BANK_6_STORED data from within the code itself, you need to have access to the map file variable `__BANK_6_tail` which is not available until after the binary is generated.  So there is no way to get a memory address for the appended section from within the compile.

aralbrec, 2017-10-01:
This case from the previous post:

```
SECTION BANK_6_STORED
org 0xc000, 0x2000  ;; apply org and phase
```

Is just extending the same model for sections we are using now -- a section can have an org or not, now we're adding a phase or not.

I think this is less useful than the above case where the section only has a phase and it's part of another section with org.  However, `appmake` will take this section binary and place it into BANK_6 at address 0xc000 so this is a viable way to do things - you could use this to org the section at a physical address and have it assemble for another address with phase.

pauloscustodio, 2017-10-01:
Thanks for the explanation, I understand the need.

Let me a couple of days to think about the easiest way to build this in.

aralbrec, 2017-10-01:
> But to get the address of the appended BANK_6_STORED data from within the code itself, you need to have access to the map file variable __BANK_6_tail which is not available until after the binary is [...]

Just to add, that variable does seem to be available at compile time.  We're doing this to initialize the data section:

```
      EXTERN __CODE_END_tail
      EXTERN __DATA_head, __DATA_END_tail

      ld hl,__CODE_END_tail
      ld de,__DATA_head
      ld bc,__DATA_END_tail - __DATA_head

      EXTERN asm_memcpy
      call   asm_memcpy
```

and that's referring to the section org and sizes.

So that manual method of concatenating the 0x2000 section to the end of BANK_6 will work and the user can grab the address from __BANK_6_tail.

Getting the address of phased sections is also strange in that a label must be present prior to the PHASE to point at the physical address or you must grab a physical address from a section as here.

feilipu, 2017-10-17:
For the Z180 CPU it would be useful to have two additional named sections, being`COMMON_AREA_0` (which always has `ORG 0`) and `COMMON_AREA_1` which can take any `ORG`.

The actual names of these two sections is irrelevant. Those are just suggestions. Perhaps `CA_0` and `CA_1` is better shorthand? But the idea is to align with the Z180 hardware MMU that provides these two memory regions.

Ideally, since I'm asking, I would like to have a separate `ORG` available for `SECTION COMMON_AREA_1_CODE` and `SECTION COMMON_AREA_1_DATA`.

The idea (which is used in a number of implementations), is to have system wide code and data in high memory `COMMON_AREA_1`, whilst the rest of the logical space, up to 60kB, is banked into `BANK_0` through `BANK15`.