z88dk/z88dk #427: putsprite failing on msx (bluemsx)

zx70, 2017-09-30:
putsprite is oddly failing on the msx target,  garbled areas are filled, looks like the sprite data pointers are lost.
The same code works well on other targets (eg Spectravideo SVI)

zx70, 2017-09-30:
I forgot to mention that the video access sequence is mostly working, eg. Chessboard.c draws the board then fails on putting the pieces on it.

zx70, 2017-10-03:
I don't know if this is related to the same problem but I observed that the data stored in the "bss_clib" section is lost (set to 0) on the SVI target, but it is correctly loaded on the MSX.
Declaring it under "data_clib" would eventually solve also for the SVI.

suborb, 2017-10-28:
Anything that's in bss_clib will be cleared on startup so that makes sense. Which data is misplaced?

zx70, 2017-10-29:
They were two different situations,  but spotting the bss declarations to check possible misuses is an easy task,  thanks for the hint.

zx70, 2017-11-01:
I think I'm hitting something serious on z80asm, I'm investigating to reduce the affected code and get a simpler test case.

zx70, 2017-11-02:
I discovered that the "serious" stability problems are again connected to the sections.
I just updated {z88dk}/libsrc/stdio/ansi/msx/f_ansi_char.asm (mostly declared as smc_clib SECTION).
To get stable results on all the involved targets I had to comment out the "section" declaration in the file bottom:
;	SECTION bss_clib	
.chline_buffer
	defs 2

On the MTX target it had the very strange effect to alter the value of the parameter passed by  {z88dk}/libsrc/stdio/ansi/msx/f_ansi_attr.asm:
        SECTION data_clib
	PUBLIC	msx_attr
.msx_attr			defb $F1	; White on Black

Oddly commenting out the section declaration above seems to avoid the problem.

zx70, 2017-11-02:
..could it be that on some target the data_clib and the bss_clib sections are overlapped ?

suborb, 2017-11-02:
Have you got an example that I can compile and look at the .map file of?

For RAM compiles, the order is rodata, smc, data, bss, for ROM it's rodata (in ROM), then in RAM bss, smc, data.

zx70, 2017-11-02:
ansitest.c:
zcc +mtx -subtype=mtx -lndos -create-app -clib=ansi  -startup=2 -pragma-define:ansicolumns=64 -Ca--map ansitest.c
I will try it too, but first the commented out section declaration must be restored.

zx70, 2017-11-02:
__sgoioblk                      = $598D ; addr, public, , mtx_crt0, bss_crt, ../crt0_section.asm:133
__sgoioblk_end               = $59C9 ; addr, public, , mtx_crt0, bss_crt, ../crt0_section.asm:134

msx_attr                        = $598A ; addr, public, , f_ansi_attr, data_clib, stdio/ansi/msx/f_ansi_attr.asm:162

_errno                            = $598B ; addr, public, , _errno, bss_error, .....z80/_errno.asm:7

__smc_clib_tail                 = $598A ; const, public, def, , ,
__smc_user_head                 = $598A ; const, public, def, , ,
__smc_user_tail                 = $598A ; const, public, def, , ,
__data_clib_head                = $598A ; const, public, def, , ,

__data_clib_tail                = $598B ; const, public, def, , ,
__data_crt_head                 = $598B ; const, public, def, , ,
__data_crt_tail                 = $598B ; const, public, def, , ,
__data_compiler_head            = $598B ; const, public, def, , ,
__data_compiler_tail            = $598B ; const, public, def, , ,
__data_user_head                = $598B ; const, public, def, , ,
__data_user_tail                = $598B ; const, public, def, , ,
(...)

suborb, 2017-11-02:
Maybe I'm getting something different from you - I'm using memu.

Regardless of where the msx_attr variable is located, nastiest runs correctly, but on exit ends up with a whole bunch of gibberish being output to the screen.

zx70, 2017-11-02:
Ok,  it is now clear that removing BSS section declaration was simply permitting the last defs to be part of the SMC section and act as a workaround.

A fully functioning workaround:

Section smc_clib
(...)
nop
nop
SECTION bss_clib
.chline_buffer
defs 2

zx70, 2017-11-02:
I'm now investigating the putsprite problem on the msx target in a similar way.

zx70, 2017-11-02:
I use memu too, better to Clean up my current z88dk tree and work on a cleaner release

zx70, 2017-11-02:
Ansitest clearly failed, here..

zx70, 2017-11-02:
Example taken from clock.c
======================

Sprite data position  on MSX:
_roman_nums                     = $AFCE ; addr, public, , clock, data_compiler, clock.c:18
__rodata_clib_tail              = $AFCE ; const, public, def, , ,
__rodata_user_head              = $AFCE ; const, public, def, , ,
__rodata_user_tail              = $AFCE ; const, public, def, , ,
__rodata_font_head              = $AFCE ; const, public, def, , ,
__rodata_font_tail              = $AFCE ; const, public, def, , ,
(...)
__data_clib_head                = $AFCE ; const, public, def, , ,
__data_clib_tail                = $AFCE ; const, public, def, , ,
__data_clib_size                = $0000 ; const, public, def, , ,
__data_crt_head                 = $AFCE ; const, public, def, , ,
__data_crt_tail                 = $AFCE ; const, public, def, , ,
__data_crt_size                 = $0000 ; const, public, def, , ,
__data_compiler_head            = $AFCE ; const, public, def, , ,

Sprite data on SVI:
exitsp                          = $9B9A ; addr, public, , svi_crt0, bss_crt, c:/z88dk/lib/config\..\..\/lib/crt0_section.asm:142
exitcount                       = $9B9C ; addr, public, , svi_crt0, bss_crt, c:/z88dk/lib/config\..\..\/lib/crt0_section.asm:143
_roman_nums                     = $9A95 ; addr, public, , clock, data_compiler, clock.c:18
_x                              = $9B9D ; addr, public, , clock, bss_compiler, clock.c:1290
_y                              = $9B9F ; addr, public, , clock, bss_compiler, clock.c:1291

__smc_user_head                 = $9A93 ; const, public, def, , ,
__smc_user_tail                 = $9A93 ; const, public, def, , ,
__smc_user_size                 = $0000 ; const, public, def, , ,
__data_clib_head                = $9A93 ; const, public, def, , ,
__data_clib_tail                = $9A95 ; const, public, def, , ,
__data_clib_size                = $0002 ; const, public, def, , ,
__data_crt_head                 = $9A95 ; const, public, def, , ,
__data_crt_tail                 = $9A95 ; const, public, def, , ,
__data_crt_size                 = $0000 ; const, public, def, , ,

zx70, 2017-11-02:
On the SVI the different data_clib_size may be saving it from crashing?

aralbrec, 2017-11-02:
Got some bugs in compiles from the recent sccz80 changes too, maybe check on that as well.

suborb, 2017-11-02:
On the MSX it seems to use graphics/putsprite2 but on the svi it's graphics/msx/putsprite is that right?

suborb, 2017-11-02:
> Got some bugs in compiles from the recent sccz80 changes too, maybe check on that as well.

Can you let me know what they are and I'll fix them up/get some testcases in there.

aralbrec, 2017-11-03:
I've opened a couple of issues

zx70, 2017-11-03:
Dom, I found the putsprite2 error only this morning and then I saw your comment.    Yes,  it is the wrong file, I must have forgotten it in the list after some experiment, sorry !   :(

zx70, 2017-11-03:
I'll play a bit more around te mtx before closing the issue,  but I must thank you for the support!

zx70, 2017-11-04:
I'm taking too long with my extra experiments on the MTX and the bug connected to this topic is now solved,  I can open another issue if refreshing my z88dk copy won't help.

closed 2017-11-04 by zx70