z88dk/z88dk #433: cpcrslib freezes

Fabrizio-Caruso, 2017-10-03:
I am trying to use cpcrslib with Z88DK. I am getting freezes anytime I try to 
either set video mode or select the colors.
I have tried both firmware and non-firmware routines from cpcrslib but the result
is either a freeze or no effect. I do disable the firmware before using the non-firmware routines.
So far I have been able to use cpcrslib to print characters on the screen (may 10-20 faster than with conio.h).

I want to use mode 1 (320x200 and 4 colors)
I am currently stuck with the four standard colors and I would like to use different ones.
Is there any other way to set the 4 colors in mode 1 by maybe using directives?

zx70, 2017-10-05:
As clarified in the direct chat the cpcsrlib does not seem to rely on the current z88dk firmware interposer, or some similar update is required.    Fabrizio solved by using a custom bios call and the mentioned interposer .

zx70, 2017-10-07:
It should be resolved now,  I suggested to use the firmware interposer to call the rom.

closed 2017-10-07 by zx70