z88dk/z88dk #439: appmake doesn't append *_DATA.bin when org_data is set

feilipu, 2017-10-07:
Within the last few commits appmake has ceased to correctly append the DATA binary at the correct position, when the `TAR__crt_org_data` is set to a specific address.

Tested with the password example program and following command lines.
```bash
zcc +rc2014 -v -subtype=basic_dcio -SO3 -clib=sdcc_iy --max-allocs-per-node200000 password.c -o password -create-app
zcc +yaz180 -v -subtype=basic_dcio -SO3 -clib=sdcc_iy --max-allocs-per-node200000 password.c -o password -create-app
```
In the resulting Intel HEX file, the DATA section is missing.
[password org_data 0xA000 rc2014](https://github.com/z88dk/z88dk/files/1364665/password0xA000rc2014.txt)
[password org_data 0 rc2014](https://github.com/z88dk/z88dk/files/1364664/password0rc2014.txt)

feilipu, 2017-10-07:
Sorry. User error. Nothing to see here.
Use `appmake +glue --hex` to get the outcome. `zcc -create-app` has never done this.

Command line for reference, if the `*_DATA.bin` (and `*_BSS.bin`) file is not appended to the `*_CODE.bin` file.
```bash
$ zcc +yaz180 -v -m -subtype=basic_dcio -SO3 -clib=sdcc_iy --max-allocs-per-node200000 password.c -o password
$ appmake +glue -b password --clean --ihex
```

closed 2017-10-07 by feilipu

aralbrec, 2017-10-07:
You should get a warning from `appmake +glue` for rom model compiles.  This is because rom model compiles require the data section to be appended to the code section in either compressed or uncompressed form so that the data section can be initialized in ram at startup.  `appmake +glue` doesn't currently do that and requires the user to do it.

`appmake +rom`, which `-create-app` is using for the rc2014, does take care of this automatically for rom model compiles.  For a ram model compile, the default is to append DATA to CODE and there should only be a single *_CODE.bin binary generated (and possible separate *_BSS.bin which is ignored and can be ignored if the crt is told to zero the bss section via pragma or default setting).  If you've changed the DATA org away from 0 it should be generating a separate *_DATA.bin too and `appmake +rom` will not put the data section into the output as it assumes the *_CODE.bin file contains everything.  You have to add the data section yourself manually by using `appmake +ihex` directly.  I suppose we could try to do correct behaviour for this but that crosses into `+glue`'s functionality where this is all taken care of properly for any set of section binaries, not just CODE, DATA and BSS.