z88dk/z88dk #448: CPC firmware routine access [help]

Fabrizio-Caruso, 2017-10-09:
I am trying to use sound queues in CPC to get some sound. So far no sound or random crackling sounds even if I use supposedly working examples from the CPCwiki forum:
http://www.cpcwiki.eu/index.php/Programming:Tutorial_-_Understanding_the_fundamentals_of_BASIC_SOUND_and_the_Firmware_SOUND_QUEUE

For instance:
The BASIC command sound 1,42,50
should be equivalent to
void __ping_sound(void)
{
#asm
	EXTERN firmware
	
	jr ping_code
	
	ping_data: 
		defb 1,0,0,142,0,0,12,50,0  
		
	ping_code:
		ld hl,ping_data
		call firmware
		defw 0xbcaa
#endasm	
}

but the ping function produces no sound at all.
I wonder what I am doing wrong...

Fabrizio-Caruso, 2017-10-10:
I don't know what is wrong. I am using some Assembly code that should produce some sounds.
I wonder whether the problem is the interposer or something else.

Ideally it would be nice to have some c function to use sound queues (CPC firmware routines) and write onto PSG registers as done for the MSX target that has the very same sound chip.

zx70, 2017-10-12:
I just got a hint from Fabrizio, fix will follow soon

zx70, 2017-10-13:
fabrizio is now using internal z88dk functions for psg sounds,  issue closed.

closed 2017-10-13 by zx70