z88dk/z88dk #450: ZX81 in HRG mode - "interpreting indirect value as immediate"

zx70, 2017-10-10:
zcc +zx81 -subtype=wrx -clib=ansi -create-app  ansitest.c

Warning at file 'c:/z88dk/lib/config\..\..\/lib/zx81_hrg.def' line 243: interpreting indirect value as immediate
                   ^ ----         call    (HRG_LineStart + $8000)

I'm not understanding what is the error about, HRG_LineStart is inside the same module (zx81_crt0), not split into sections.

zx70, 2017-10-10:
Looks like removing the parentheses solves the problem.

closed 2017-10-10 by zx70

pauloscustodio, 2017-10-10:
The warning is to signal that the user might want to use an indirect value (NN) where one is not allowed, and the value is interpreted as immediate NN.

The warning disappears wither by removing the parentheses or writing a unary plus +(NN).

zx70, 2017-10-10:
Got it, it looked like an error but it is absolutely appropriate.