z88dk/z88dk #457: Not building at the moment (13.10.17)

maziac, 2017-10-14:
On Linux a git clone and ./build.sh ends up in

cc -MMD -I. -Ilib -It -g -Wall -ftabstop=4 -std=gnu11 -O3 -I../../ext/uthash/src -g -O2 -c -o errors.o errors.c
In file included from errors.c:13:0:
lib/fileutil.h:16:21: fatal error: utarray.h: No such file or directory
 #include "utarray.h"
                     ^
compilation terminated.

The utarray.h file seems to have vanished.

suborb, 2017-10-14:
Utarray has moved into a sub module along with some other 3rd party libraries. So the update procedure is slightly different:

To clone with submodules: git clone --recursive https://github.com/z88dk/z88dk.git 

To expand the cloned directory tree: git submodule update --init --recursive

aralbrec, 2017-10-14:
The nightly source package does not include the submodules so the build instructions will have to be updated:
https://www.z88dk.org/wiki/doku.php?id=temp:front#installation

The front page readme.md should probably be updated too - the green clone button (zip or git clone) does not fetch the submodules either.

suborb, 2017-10-14:
The nightly tar/zip balls do include the submodules - the source in those files is the same directory that's used for building with the exclusion of the .git directory.

Agree that the README needs an update to mention that the submodules need updating, I'll add some words later.

closed 2017-10-15 by suborb