z88dk/z88dk #460: SVI 318 (16k ram model) not supported by SVI despite being listed as supported

Fabrizio-Caruso, 2017-10-16:
The SVI 318 model (16k ram model) is not supported by the SVI target 
despite being listed as supported models.

Any problem (even a simple hello world) when loaded on SVI 318 (tested on BlueMSX) will no produce any effect. The same program will run fine on SVI 328.

It may just be a that the SVI 318 expects the program in a different memory location.

Fabrizio-Caruso, 2017-10-18:
I have verified the very same problem with MSX models with just 16k of RAM.

zx70, 2017-10-19:
The hardcoded ORG location is (was) 40000 on the MSX.
I just updated appmake to permit some flexibility, I noticed in the ROM basic that the bottom memory for BASIC is hardcoded at 32768, so since the 16K models are not working we may guess they use the higher half (which is probably correct, because the higher locations are used to store the system variables).
After the usual nightly build you will be able to use the "-zorg" parameter on the MSX, please try with "-zorg=49200" not to be too close to 49151.

Fabrizio-Caruso, 2017-10-20:
Problem still there on both svi and msx

zx70, 2017-10-20:
Please wait for a successi nightly build, appmake was not able to receive the zorg parameter..  or did you  pick and build the sources from bit?

Fabrizio-Caruso, 2017-10-21:
Problem still there with the today's nightly build (21st of oct) but it is actually worse because the compiled .cas is corrupted when loaded in a model with more than 16k of ram (the bigger model SVI-328 loads the game but some corruption is visible on the screen as if the code had ened up in video ram).

Fabrizio-Caruso, 2017-10-21:
Remark: This issue can be reproduced by a simple hello world program. It is unrelated to whatever my code is doing.

zx70, 2017-10-21:
The only 'fixed' target is,  at the moment,  the msx.  The svi is left untouched. 
Did you try to build a program with the new ram position, eg. setting -zorg=49200 ?

zx70, 2017-10-21:
The ideal test target is msx 16k

Fabrizio-Caruso, 2017-10-22:
Indeed! MSX 16k is fixed (tested with OpenMSX). SVI 318 (Spectravideo with 16k) is not (tested with BlueMSX)

zx70, 2017-10-22:
Code for svi should be ready for tonight.  *IF* the nightly build will succeed you'll have it by tomorrow morning.

zx70, 2017-10-24:
Fabrizio confirmed that the 16K mode is now correct also on the Spectravideo SVI

closed 2017-10-24 by zx70