z88dk/z88dk #464: [Spectrum] How to redefine characters with clib=ansi?

Fabrizio-Caruso, 2017-10-19:
I am no longer able to redefine characters with the Spectrum target since I have switched to clib=ansi.
Previously I could
- either using new lib, stropts.h and extern unsigned char font_8x8_ROM[]
- or using classic lib with no ansi and poke starting at 0xff58

Now, due to the improvements in conio.h I am using clib=ansi but writing onto 0xff58 has no longer any effect and the new lib method is not available either.

How can I redefine characters?

It would be great in the future to have a generic way to perform this common use-case.

suborb, 2017-10-19:
I think without ansi in classic there’s 3 ways to do it. That way, by control code and by a compile time pragma. 

The ansi library for zx can be configured to use a different font using a compile time pragma. The wiki suggests:

-pragma-define:ansifont=15616 -pragma-define:ansifont_is_packed=0

To switch to the rom font. So to use a different font just change the address in the pragma.

Fabrizio-Caruso, 2017-10-19:
How do I know where to put my new font set? Can I put it anywhere and the compiler relocates the code elsewhere?
If, for instance, I put it at 32768, can I copy the rom font set from 15616 to 32768?
What does start of the font set mean? Character 0 or character 32?
How should I compile my code in this case? -startup=1?

suborb, 2017-10-19:
Here's an example command line:

```
% cd examples/console
% zcc +zx -clib=ansi ansitest.c -lndos -pragma-redirect:ansifont=_font_8x8_bbc_system -pragma-define:ansifont_is_packed=0 -pragma-define:ansicolumns=32 -create-app
```

This will compile the ansi test using the BBC font that's available in the library. Other builtin fonts are:

```
_font_8x8_bbc_system
_font_8x8_clairsys
_font_8x8_clairsys_bold
_font_8x8_zx_system
```

You can define your own font of course - 768 bytes, 8 bytes per character starting at character 32.

Fabrizio-Caruso, 2017-10-19:
How do I define the font? Should I edit a file? Can I then specify the location of the new font file with -pragma-redirect:<location of new font >?

suborb, 2017-10-19:
The font should be your own .asm file, structure it like this:

```
SECTION rodata_user

PUBLIC _font_myname

_font_myname:
      BINARY "font.bin"
```

and add it to your project.

You can then refer to it via: `-pragma-redirect:ansifont=_font_myname`. 

The actually binary is in the usual format 8 bytes for character 32, 8 bytes for character 33 and so on.

Fabrizio-Caruso, 2017-10-19:
Thanks a lot!

Where can I find the _font_8x8_zx_system fonts? 
I will start from them and than manually compute the offset of each character I want to redefine.

The other methods (which do not work with clib=ansi) are easier for the coder because I could always
have my code do the math (e.g., POKE(UDG_BASE+MY_CHAR*8+i,i) ).
This method is more memory (and time) efficient, though.

suborb, 2017-10-19:
Those fonts are lifted from newlib, so they're in `libsrc/_DEVELOPMENT/font/font_8x8/`

On most targets you can do the same trick ((MY_CHAR - 32) * 8 ) + font_address to get to the data for the font.

There's probably newer tools that run on a PC, but I always used to use Utility1 (http://www.worldofspectrum.org/infoseekid.cgi?id=0008932) to edit fonts and UDGs

However, the font is nominally stored in a rodata section so on ROM based targets it will be stored in ROM.

Fabrizio-Caruso, 2017-10-19:
Thanks a lot!
I think the font file method is good despite the fact it is convenient in my case because I have all the new character definitions and I just need to know where to place them. I do not need to edit the characters because I have already done it. This method could work on multiple targets. Am I right?

A second method that should be allowed whenever possible is to let the user "poke" the memory as in BASIC. This is the natural way and people can easily find documentation with BASIC examples and do the same in C. This method works on classic no-ansi but it no longer works with ansi. Is it wanted? A bug?

Fabrizio-Caruso, 2017-10-19:
I could not find editable rom font files. The rom files are in some sort of binary format.
How can I add my character definitions, i.e., how do I modify a given charater for which I already have the 8 byte of data?
Can I poke at given memory area to modify them??

Fabrizio-Caruso, 2017-10-19:
Using a copy of the BBC font set produces errors:
Error at file 'zcc_opt.def' line 4: syntax error
Error at file 'zcc_opt.def' line 6: syntax error
2 errors occurred during assembly
Errors in source file c:/z88dk/\lib\spec_crt0.asm:
Error at file 'zcc_opt.def' line 4: syntax error
                   ^ ----       EXTERN ./src/spectrum/fonts/_font_8x8_bbc_system
Error at file 'zcc_opt.def' line 6: syntax error
                   ^ ----       defc ansifont = ./src/spectrum/fonts/_font_8x8_bbc_system
make: *** [Makefile:207: spectrum_clib_udg] Error 1

suborb, 2017-10-19:
The pragma define takes the label of the font not the file name. 

The font format is in “standard” zx font format - I detailed it briefly above. 

So to insert your characters grab a hex editor and change the 8 bytes at the offset you’ve calculated using the formulat I mentioned above.

Fabrizio-Caruso, 2017-10-19:
No simpler solution? 
No way to have my program copy at some ram locations where font is stored?
So, I need to do the manual change for all my characters... No other method.
Any hex editor you would suggest to make this process easier?
Are text human-readable files also supported?

aralbrec, 2017-10-19:
You can edit the fonts themselves and then just include the edited font in your program so that your program does not have to change the font or you can store the plain font in your program and have your program modify it at runtime.

If you're using windows, a really good font editor is the [fzx editor](http://www.zx-modules.de/) by Claus.  You'll have to click on "fzx editor" among the banners on the left and then scroll down to the "LOAD _" under download here.  That will get you the zip containing a windows exe.

This editor can do standard 8x8 fonts and fzx proportional fonts.  The few 8x8 fonts we have can be found in [libsrc/_DEVELOPMENT/font/font_8x8](https://github.com/z88dk/z88dk/tree/master/libsrc/_DEVELOPMENT/font/font_8x8), particularly the raw uncompressed binaries font_8x8_bbc_system.bin, font_8x8_clairsys.bin, font_8x8_clairsys_bold.bin and font_8x8_zx_system.bin.  These are just raw binaries, 8 bytes per character starting from space.  These files will usually be 768 bytes covering 8x8 bitmaps for ascii codes 32-127.  If you copy one of these files and change the file extension to ".chr" then the fzx editor can load it and let you edit the font.  Save the edited font and include it like usual with the BINARY directive in a .asm file.

If you're making changes at runtime instead, you have to keep in mind the difference between a rom target and a ram target.  If you're using a ram target, like the zx spectrum where the entire program is loaded into ram, then you can just poke the font directly in memory.  However if you are using a rom target, like msx cartridge, the font data is not in ram - it's in rom.  You cannot poke into rom.  To make it work you would make sure the font is in a DATA section instead of a ROM section.

Normally fonts are placed in an RODATA section (read only data) which is part of the rom:

```
SECTION rodata_font_8x8

PUBLIC _zx7_font_8x8_bbc_system
PUBLIC _zx7_font_8x8_bbc_system_end

_zx7_font_8x8_bbc_system:

   BINARY "font_8x8_bbc_system.bin.zx7"

_zx7_font_8x8_bbc_system_end:
```

Instead you would want to place the font in a data section:

```
SECTION data_user

PUBLIC _zx7_font_8x8_bbc_system
PUBLIC _zx7_font_8x8_bbc_system_end

_zx7_font_8x8_bbc_system:

   BINARY "font_8x8_bbc_system.bin.zx7"

_zx7_font_8x8_bbc_system_end:
```

This makes no difference to a ram model compile.  However, in a rom model compile, the DATA section will be stored in rom and then copied to ram at start up so that your program can then make changes at runtime.  But note there are now two copies of the font :- one in rom and the one used by the program in ram.  To help mitigate this you can compile using the compressed rom model, which will compress the stored copy of the font in rom using zx7 (an optimal lz77 compressor).  Then you will have a compressed copy of the font in rom which will be decompressed into ram where you can make runtime modifications.

However, I think it's preferable to just edit the font beforehand so that no runtime modifications need to be made.

Fabrizio-Caruso, 2017-10-19:
@aralbrec, THANKS! 
I already have the font data, which I am already using in about 10 different targets.
I do not need to re-edit them. I just need to use them in Spectrum (no rom) with clib=ansi because it is the only configuration with fast character graphics (conio.h is now fast).

I don't mind about some memory being wasted. I just want a not too painful method for me (I have already implemented this for clib=sddc_y and classic lib with no luck due to slow character graphics with these configs).

Where should I write?
SECTION data_user

PUBLIC _zx7_font_8x8_bbc_system
PUBLIC _zx7_font_8x8_bbc_system_end

_zx7_font_8x8_bbc_system:

   BINARY "font_8x8_bbc_system.bin.zx7"

_zx7_font_8x8_bbc_system_end:

A file that I should link? What should I call the file with the above content?

How do I compile?
And how do modify the font in RAM? Which address?

aralbrec, 2017-10-19:
If you have your own font, you can add a short asm file to the project to add it to the compile:

myfont.asm
```
SECTION data_user   ;; change to rodata_user if you will not be modifying it at runtime

PUBLIC _myfont

_myfont:

   BINARY "myfontdata.bin"   ;; the raw binary containing your font data
```

Then add "myfont.asm" to the compile line.

In your C code you can add this:

```
extern unsigned char myfont[];
```

Notice "myfont" will point at the public symbol "_myfont" because the c compiler mangles names by adding a leading underscore.  This declaration treats the font data like a character array.  So you can just poke into it with `myfont[100]=0;` etc.

To make the clib=ansi driver use this font, you add to the compile line as suborb mentioned:
`-pragma-redirect:ansifont=_font_myname`

BTW, if you have a lot of pragma defines, pragma reidirects and whatnot, you can put them into a file instead that is read by zcc if you add `-pragma-include:zpragma.inc`  (zpragma.inc being the filename)

Fabrizio-Caruso, 2017-10-19:
How do I tell the compiler where the binary file is? I just want an exact copy of the Spectrum rom font set, which I would like to modify through POKEs.

I would like to make my life simple since I am writing a game for 50-60 targets (across multiple tool kits where Z88DK is one of the two most important ones).
Ideally I would like to have a file containing:
	static const char player_down[8] =      { 24, 36, 24,102,153, 24, 36,102};
	static const char player_up[8] =        { 24, 60, 24,102,153, 24, 36,102};
	static const char player_right[8] =     { 24, 52, 25,118,152, 24, 20, 20};	
	static const char player_left[8] =      { 24, 44,152,110, 25, 24, 40, 40};
	static const char ghost[8] =            {129,126,165,129,129,189,129,126};
	static const char missile_right[8] =    {  0,  0, 15,252,252, 15,  0,  0};
	static const char missile_left[8] =     {  0,  0,240, 63, 63,240,  0,  0};
	static const char invincible_ghost[8] = { 60, 66,165,129, 90, 36, 36, 60};
	static const char gun[8] =              {  0,128,126,200,248,192,128,  0};
	static const char powerUp[8] =          {  0, 60, 54,223,231,122, 36, 24};
	static const char missile[8] =          {  0,  0,  8, 56, 28, 16,  0,  0};
	static const char bomb[8] =             { 60, 66,165,153,153,165, 66, 60};
	static const char bubble[8] =           { 24, 60, 60, 60,126, 90, 66, 66};
	static const char invincibility[8] =    { 24, 36, 24,  0,153,  0, 36,102};
	static const char vertical_brick[8] =   { 24, 24, 24, 48, 24, 12, 24, 24};
	static const char horizontal_brick[8] = {  0,  0,  0,255,  0,  0,  0,  0};	
        // more stuff later

which I want to re-use all multiple targets across multiple tool kits. Wasting 1k or 2k of ram is not an issue for my game.

aralbrec, 2017-10-19:
The fonts are in the library, however they're in an rodata section which is not what you want so you have to include the font binary directly to place it in the data section:

```
SECTION data_user

PUBLIC _font

_font:

   BINARY "${ZCCCFG}/../../libsrc/_DEVELOPMENT/font/font_8x8/font_8x8_zx_system.bin"
```

That's what it would be but z80asm does not expand environment variables in filenames but maybe it should @pauloscustodio ?

So at the moment you will have to copy that file to your own project directory so that you can do this:

```
SECTION data_user

PUBLIC _font

_font:

   BINARY "font_8x8_zx_system.bin"
```

aralbrec, 2017-10-19:
To compactly redefine some characters in the font at runtime, an array that associates character code with bitmap may help:

```
extern unsigned char font[];

struct redefine
{
   unsigned char ascii;
   unsigned char bitmap[8];
};

struct redefine map[] = {
   'g',  { 0,128,126,200,248,192,128, 0},   // gun
...
};

for (i = 0; i < sizeof(map) / sizeof(*map); ++i)
   memcpy(font + map[i].ascii*8, map[i].bitmap, 8);
```

Fabrizio-Caruso, 2017-10-19:
The compiler does not find the files...

I fear the compiler looks in its own directories...

/cygdrive/c/z88dk/bin/zcc.exe +zx -D -clib=ansi  -pragma-define:ansicolumns=32 -vn -DFULL_GAME -DREDEFINED_CHARS -DSOUNDS -DCLIB_ANSI  -D__SPECTRUM__ -lndos -create-app -o build/FULL_spectrum_48k_udg_clib.prg src/spectrum/udg.asm src/spectrum/spectrum_graphics.c src/display_macros.c src/enemy.c src/invincible_enemy.c src/level.c src/character.c src/text.c src/missile.c src/strategy.c src/input_macros.c src/main.c
sccz80:"src/spectrum/spectrum_graphics.c" L:105 Warning:#9:Converting pointer to integer without cast
Error at file 'src/spectrum/spectrum_graphics.c' line 424: symbol '_udf' not defined
1 errors occurred during assembly
Errors in source file c:/z88dk/\lib\spec_crt0.asm:
Error at file 'src/spectrum/spectrum_graphics.c' line 424: symbol '_udf' not defined
                   ^ ---- (null)make: *** [Makefile:207: spectrum_clib_udg] Error 1

Fabrizio-Caruso, 2017-10-19:
THANKS EVERYONE! 
I have managed to redefine characters with run time writes into memory.
I may switch to the more efficient solution with no useless copying of data at run-time.

Sorry for asking many questions... I am a bit confused by the so many different solutions to do the very same thing.
For the sake of finding a solution for smooth color character graphics on the Spectrum I have fully implemented the graphics of my game using:
- clib non-ansi native directives + poke to change fonts
- newline different native directives + copy at run time + stropt.h 
- clib ansi with conio.h (old and new versions) + the latest approach...

Nevertheless I am happy because my game now has smooth graphics on the Spectrum, which at first was VERY slow and I was losing hope.

closed 2017-10-19 by Fabrizio-Caruso