z88dk/z88dk #468: +z180: make default memory map consistent with z180 banking scheme

aralbrec, 2017-10-22:
Moving this from #425 so it's not lost.

> For the Z180 CPU it would be useful to have two additional named sections, beingCOMMON_AREA_0 and COMMON_AREA_1.
> 
> The actual names of these two sections is irrelevant. Those are just suggestions. Perhaps CA_0 and CA_1 is better shorthand? But the idea is to align with the Z180 hardware MMU that provides these two memory regions.
> 
> Ideally, since I'm asking, I would like to have a separate a ORG available for COMMON_AREA_1_CODE and COMMON_AREA_1_DATA, with the COMMON_AREA_1_DATA being compressible, and both areas being copied into their correct location by the _crt_init or _crt_start code.
> 
> The idea (which is used in a number of implementations), is to have system wide code and data in high memory COMMON_AREA_1, whilst the rest of the logical space, up to 60kB, is banked into BANK_0 through BANK15.

Brief answer for now:  I was originally looking at doing something like this but I decided against it at that time because the z180 banks can move so there is no fixed address for anything.  Many parallel processes or even a single program may be modifying the memory map to access specific things from the larger memory space while moving the location of common1 and the bank area as necessary.  With things movable like that I thought it better if the user just defined the bank equivalents with org for various common_1, etc, and manage everything according to needs.  So that's why the z80 map is still there for the z180.

It is also possible to add code to section code_crt_init so that other data/bss sections are initialized before main starts at compile time.

feilipu, 2017-10-22:
Thanks for making this a separate issue.
I'm working around this by doing following.

Adding three sections to end of the `CODE` section in a `crt_yabios_memory_model_z180.inc` memory map.
```asm
;...

section rodata_page0
section rodata_common1_data
section rodata_common1_driver

SECTION CODE_END
```
I have put all of the contents of these three `section` into three matching files, to avoid the `PHASE` limitations, and marked them with their own destinations.
For example.
```asm
SECTION rodata_common1_data
PHASE   __COMMON_AREA_1_PHASE_DATA
;------------------------------------------------------------------------------
; start of common area 1 page aligned data
;------------------------------------------------------------------------------
```
And initialising the sections in the `crt_preamble.asm`.
```asm
    EXTERN __rodata_common1_data_head
    EXTERN __rodata_common1_data_size

    ld hl,__rodata_common1_data_head
    ld de,__COMMON_AREA_1_PHASE_DATA
    ld bc,__rodata_common1_data_size

    EXTERN asm_memcpy
    call   asm_memcpy

    EXTERN __rodata_common1_driver_head
    EXTERN __rodata_common1_driver_size

    ld hl,__rodata_common1_driver_head
    ld de,__COMMON_AREA_1_PHASE_DRIVER
    ld bc,__rodata_common1_driver_size

    EXTERN asm_memcpy
    call   asm_memcpy
```
Seems to work well enough for now (though I'm wrangling other issues of my own making).

The only loss is that the `__rodata_common1_data` is not compressed, although that is not a major issue, really. Perhaps one day it may become something to solve.

feilipu, 2017-11-17:
@aralbrec please check hotmail.

feilipu, 2018-02-02:
The comment [above](https://github.com/z88dk/z88dk/issues/468#issuecomment-338516602) has resolved this issue for me. Note it is still "clunky" in that the asm to go in each of the sections has to be in one file because of the limitation in `PHASE` offsets.

But it works well enough that I've had no problem for the past month.
Unless there are other reasons, such as #425, please close this issue.

feilipu, 2018-03-07:
This issue can be closed, based on above comment.

The only remaining limitation is within `PHASE`, and the requirement to keep all the code relating to the same `PHASE` in the same file.

feilipu, 2019-02-11:
Close. Complete.

closed 2019-02-11 by suborb