z88dk/z88dk #474: SDCC project planning a 3.7.0 release. Anything that z88dk considers important to go in before?

spth, 2017-10-26:
The SDCC project is again preparing for a SDCC 3.7.0 release (there was an earlier, failed attempt in summer). While it is not clear if or when it will happen, I wonder if there are any small changes (it might be too late for big ones by now) in SDCC that z88dk needs, that should go in before a 3.7.0 release.

There currently are still two open --reserve-regs-iy bugs:
https://sourceforge.net/p/sdcc/bugs/2370/
https://sourceforge.net/p/sdcc/bugs/2674/
but in general, SDCC seems to do much better with --reserve-regs-iy than at the 3.6.0 release.

Philipp

aralbrec, 2017-10-26:
There are a couple of smaller items that are percolating:

#156 sdcc has --constseg and --codeseg to change areas that the const data and code are assigned to.  We're missing similar options for bss and data, maybe --bssseg and --dataseg.

I see #55 too but I think we will head a different way here.  We may have targets with their own rules in addition to user-supplied rules.  I'm not sure what we will do yet - maybe construct a rule set at compile time so that there is one used by sdcc.  The requirement is not yet well thought out.

spth, 2017-11-07:
What's the difference between bss and data? All (global, static) variables are initalized, either to 0 or to whatever the programmer specified. And local variables go to the stack / registers.
sdcc z80 currently uses _DATA for what gets intialized to 0, and _INITIALIZED for the rest.

Philipp

suborb, 2017-11-07:
Similar to sdcc, the rest of z88dk (libs, sccz80, crt0) make a distinction between default initialised (bss) and explicitly initialised (data).

Unfortunately in sdcc it doesn't look like we can specify the names of these two sections which means that any non-const variables are incorporated into the main binary rather than contained in the bank.

spth, 2017-11-07:
I see, so the confusion just came from the different names. I've open a feature request (https://sourceforge.net/p/sdcc/feature-requests/544/), and will see if it can be implemented partially or fully before SDCC 3.7.0.

Philipp

spth, 2017-11-07:
If you want to place variables in banks, you might also want to have a look at the support for named address spaces in SDCC.

Philipp

spth, 2017-11-08:
For the zero-initialized data, there now is --dataseg. Nothing for the nonzero-initialized stuff yet.

Philipp

aralbrec, 2017-11-08:
> If you want to place variables in banks, you might also want to have a look at the support for named address spaces in SDCC.

Yes that's in the future too.  Right now we're just getting the file-granularity section assignment sorted.  The next step is bank assignment within a file as can be done in assembler now.  I have glanced at the embedded technical report with bank placement of variables as sdcc is doing but we're also considering special pragmas that can push and pop codeseg, constseg, dataseg, bssseg, etc.

aralbrec, 2018-05-29:
> For the zero-initialized data, there now is --dataseg. Nothing for the nonzero-initialized stuff yet.
@spth

aralbrec, 2018-05-29:
> For the zero-initialized data, there now is --dataseg. Nothing for the nonzero-initialized stuff yet.

@spth Is there any fix for the non-zero initialized data section assignment on the horizon?

spth, 2018-05-29:
SDCC developers seem to be all busy with other stuff this summer. We might to do a SDCC 3.8.0 release (this time with binaries) later this year. If bugs are found at the code generation or peephole optimizer stage, I'll try to find the time to fix them, but I don't think there will be further new features.

Posting a patch to the patch tracker at Sourceforge is probably the fastest way to get new features into SDCC now (but it might still take a bit of time to review the patch).

Philipp

feilipu, 2019-02-11:
Implemented? Obsolete?

closed 2019-02-11 by suborb