z88dk/z88dk #475: appmake +glue: allow multiple map files on command line

feilipu, 2017-10-26:
As part of writing a user _application_ that needs to reference pre-existing z88dk _library_ ROM code, I think it would be useful to allow `appmake` to accept multiple map files.

In its simplest form, the additional needed symbols, that can be found within the system ROM for example, could be concatenated onto the single application generated map file that is being accepted now.

But, it would be _nice_to_have_ to be able to name an existing map file on the command line (associated with a specific system or bios build, for example), without having to concatenate it onto the application map file each time `appmake` is run.

aralbrec, 2017-10-26:
I think any externally supplied symbols will be part of the compile so that they should be in the generated map file that way.

For example if you're generating a binary to link against, you'll be making a list of global entry points / symbol names maybe using -gp[f] to get them in defc form.  Then to actually use the entry points or symbol names, you would either add that file to the compile or you would compile that file into a library to link against.

The latter library option will only list library entry and symbols in the map file that the program uses and not necessarily all the them.

I think this is what you are referring to?

feilipu, 2017-10-27:
Yes. the `zcc ... -gp` option is exactly the outcome I'm looking for.
Clearly, someone else thought of it first. ;-)

closed 2017-10-27 by feilipu