z88dk/z88dk #479: Make problem on the MSX ans SVI targets

zx70, 2017-10-28:
The following sequence should be compiled specifically for the MSX and SVI targets.
The 'test' target should be the default for the remaining ones

make[1]: Entering directory 'c:/z88dk/libsrc/msx'
zcc +test -O2 -vn -Wn43  -c msx_blit.c
zcc +test -O2 -vn -Wn43  -c msx_blit_fill_vram.c
zcc +test -O2 -vn -Wn43  -c msx_blit_ram_vram.c
zcc +test -O2 -vn -Wn43  -c msx_calculate_side.c
zcc +test -O2 -vn -Wn43  -c msx_put_sprite_16.c
zcc +test -O2 -vn -Wn43  -c msx_put_sprite_8.c
zcc +test -O2 -vn -Wn43  -c msx_set_char.c
zcc +test -O2 -vn -Wn43  -c msx_set_char_attr.c
zcc +test -O2 -vn -Wn43  -c msx_set_char_color.c
zcc +test -O2 -vn -Wn43  -c msx_set_char_form.c
zcc +test -O2 -vn -Wn43  -c msx_set_mangled_mode.c
zcc +test -O2 -vn -Wn43  -c msx_set_sprite_16.c
zcc +test -O2 -vn -Wn43  -c msx_set_sprite_8.c
zcc +test -O2 -vn -Wn43  -c msx_set_sprite_mode.c
zcc +test -O2 -vn -Wn43  -c msx_vfill_v.c
make[1]: Leaving directory 'c:/z88dk/libsrc/msx'

zx70, 2017-10-28:
.. can we use '+$(TARGET) in msx/Makefile ?
While we're on it,  is it possible to move the rs232 libraries away from any single clib?

suborb, 2017-10-28:
> thread[2].>

Yes, take a look at stdlib/Makefile for how to do it.

They are there out of convenience that's all, trying to get as much
common stuff as possible, feel free to move the call back onto the
actual port.

Links:

  1. https://github.com/z88dk/z88dk/issues/479#issuecomment-340167168
  2. https://github.com/

zx70, 2017-10-28:
I'm  tempted to port all the stuff into conditional asm, I'll take a deeper look.

zx70, 2017-10-29:
I found a nearly perfect equivalence in the games folder.    I think we still depend on the pre - built libraries for most of the targets which is not ideal. 
Do you think we should begin where possible to force the test target?   Many of the funtions can be built correctly also by forcing the target specific definitions and trick the header files to pass the correct constants,  eg.  "-D__MSX__" etc..

zx70, 2017-10-30:
I chose to restrict the build to C modules only in three different flavors, MSX, SVI and "test" (generic).
It should also save a (very little) bit of compilation time.   
I have the impression that this library part could be also built by SDCC, but it is another story  :)

zx70, 2017-11-01:
Solved, thanks Dom!

closed 2017-11-01 by zx70