z88dk/z88dk #481: (sccz80) Optimisations for z180 multiplication

suborb, 2017-10-28:
We should be able to:

* [ ] Replace l_mult (see #312)
* [ ] Have a special case for char * char to use alt instruction

spth, 2017-11-07:
Is it really work adding backends for all the z80-related architectures in scc? It seems like duplicating work already done in sdcc to me.

Philipp

aralbrec, 2017-11-07:
The c library is much larger and much more complete in z88dk.  It's also all written in assembler which makes it much faster and smaller too.  There is a different aim in this project; whereas sdcc is cross-platform and carries a library written in C to enable that, z88dk is focused on Zilog-related cpus and is concerned with performance and enhanced libraries to satisfy platform-specific needs.  There is not just z80, z180 there are variations for different z80,z180 machines.

For integer math specifically, you can see which functions are needed by the compilers and library here:
https://github.com/z88dk/z88dk/tree/master/libsrc/_DEVELOPMENT/math/integer/small

We go up to 64x64 multiply and divide to support sdcc's 64-bit type.  Written in asm, it's much faster than sdcc's c implementation for these operations (one of the benchmarks pi.c shows the difference it makes for 32-bit math).  We also have an alternate fast math implementation that does loop unrolling and leading zero elimination that will be faster if the user is willing to pay in more memory.

I don't think we have a problem if sdcc wants to borrow some of these sorts of functions but the library as a whole is not really something that can be incorporated into sdcc.

spth, 2017-11-07:
Yes. But doesn't the zsdcc fork allow you to do all (library, ect) that, without the need for z180, etc support in scc?

Philipp

aralbrec, 2017-11-08:
I wasn't sure if scc was sdcc with typo or sccz80.  Both compilers are being supported and both share the same library.  Doing library functions for one means the other gets it too.

sccz80 is considerably faster to compile -- I have several programs that take 20-45 minutes with zsdcc and optimization high (and this is what I want when using zsdcc as I'm usually going for best code) whereas sccz80 might take half a minute.  So sccz80 is often being used as development compiler; I know of several people who are using it that way now.  There used to be a difference in code size too, but that has largely gone away with the peephole set we are using with zsdcc; it will still show if the program uses a lot of float or 32-bit math.

I don't see us discontinuing sccz80 but rather supporting as many compilers as come along, including possibly other languages.  We've already got your clang->zsdcc tool supported now, for example, however that still requires manual intervention to fix up code between the clang->zsdcc step.  There are a couple of other efforts to do a compiler on llvm but they are looking like vapourware.

feilipu, 2019-02-25:
I think that this is all finished, except perhaps the `char * char` case.
Is that correct?

suborb, 2019-02-25:
It looks like I did uchar * uchar in sccz80 so I think this one can be closed now.

closed 2019-02-25 by suborb