z88dk/z88dk #482: sccz80: bug in -O2 peephole set

aralbrec, 2017-10-29:
Placing here for the first person that gets a chance to find it.

zzz.c
```
#include <stdint.h>

int16_t foo(int16_t a, uint8_t x, uint8_t y)
{
	uint8_t d = x - y;
	a -= (int16_t)d;
	return a;
}
int16_t foo(int16_t a, uint8_t x, uint8_t y)
{
	int16_t d = x - y;
	a -= d;
	return a;
}
```

zcc +zx -vn -a -O2 zzz.c

In the generated asm for the first function:

```
	ld	hl,4	;const
	add	hl,sp
	ld	e,(hl)
	ld	h,0   ;; <<--- wrong, should be register D
	pop	hl
	and	a
	sbc	hl,de
	pop	de
	call	l_pint
```

This code is for the statement `a -= (int16_t)d;` and `HE` is being loaded with the value of variable `d` instead of register `DE`.

The problem is not present at -O1 level where the generated code is:

```
	ld	hl,4	;const
	add	hl,sp
	ld	l,(hl)
	ld	h,0
	pop	de
	call	l_sub
	pop	de
```

So there's a rule in -O2 that is eliminating this call to `l_sub` but setting the wrong registers.

suborb, 2017-10-29:
z80rules.1:2634

suborb, 2017-10-29:
I've fixed this in my sccz80_declv2 branch - I spotted there was an erroneous sign extend in the auto variable initialisation as well.

closed 2017-10-30 by suborb