z88dk/z88dk #487: Are the --reserve-regs-iy bug workarounds still needed?

spth, 2017-10-31:
Originally, I posted this to the z88dk-dev list a few days ago, but it got no responses; I guess these days the GitHub issue tracker is the preferred way of communicating with z88dk developers.

Dear z88dk developers,

in the past, there were quite a few bugs in SDCC when using
--reserve-regs-iy. In current SDCC 3.6.9 #10136 these should be fixed,
and the generated code should be nearly as bug-free as when not using
--reserve-regs-iy.

Could you please check if you see any remaining --reserve-regs-iy issues?

You probably can drop the peephole rule workarounds that z88dk used to
work around the SDCC bugs.

Philipp

suborb, 2017-10-31:
Sorry Philipp, I saw the message on the forum, but I can't answer, @aralbrec has been mainly doing the  sdcc work. He's been unusually quiet lately, so he may have new hardware to play with that's taking his time up.

aralbrec, 2017-11-01:
Yes sorry, I've been quite busy the last several weeks and will continue to be for a while at least.

The rules we have been using to correct sdcc compile issues (most to do with reserve-regs-iy) can be seen here:

feilipu, 2019-02-11:
Obsolete issue.

closed 2019-02-11 by suborb