z88dk/z88dk #490: sccz80: errors in some example programs

aralbrec, 2017-11-03:
The rewrite is not compiling some example programs correctly.  I'll list them in this thread as I find them.

[sudoku.c](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/EXAMPLES/sudoku.c)
```
zcc +zx -vn -startup=1 -O3 -clib=new sudoku.c -o sudoku -create-app

sccz80:"sudoku.c" L:91 Warning:Pointer arithmetic with non-matching types: struc
t sudoku_cell *sc-struct sudoku_cell grid[81]
sccz80:"sudoku.c" L:92 Warning:Pointer arithmetic with non-matching types: struc
t sudoku_cell *sc-struct sudoku_cell grid[81]
sccz80:"sudoku.c" L:153 Warning:Pointer arithmetic with non-matching types: stru
ct sudoku_cell *sc1-struct sudoku_cell grid[81]
sccz80:"sudoku.c" L:154 Warning:Pointer arithmetic with non-matching types: stru
ct sudoku_cell *sc1-struct sudoku_cell grid[81]
sccz80:"sudoku.c" L:154 Warning:Pointer arithmetic with non-matching types: stru
ct sudoku_cell *sc2-struct sudoku_cell grid[81]
sccz80:"sudoku.c" L:155 Warning:Pointer arithmetic with non-matching types: stru
ct sudoku_cell *sc2-struct sudoku_cell grid[81]
sccz80:"sudoku.c" L:338 Warning:Control reaches end of non-void function
```

The warnings are incorrect - the array is same type for pointer arithmetic.  The program compiles but produces incorrect results.  You can try with this sudoku:

```
790000300
000006900
800030076
000005002
005418700
400700000
610090008
002300000
009000054
```

aralbrec, 2017-11-03:
[open_memstream.c](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/EXAMPLES/open_memstream.c)

```
zcc +zx -vn -O3 -clib=new open_memstream.c -o openmem -create-app --list

sccz80:"open_memstream.c" L:30 Warning:Loss of precision, converting int  to lon
g offset
```

This isn't a loss of precision since you're going from 16-bit to 32-bit.  But anyway the code generated connected to this warning is not right:

```
71    005F  21 04 00    	ld	hl,4	;const
72    0062  CD 00 00    	call	l_gintspsp	;
73    0065  21 00 00    	ld	hl,0	;const
74    0068  D5          	push	de   ;;; <- wrong, de is not 0
75    0069  E5          	push	hl
76    006A  21 00 00    	ld	hl,0	;const
77    006D  E5          	push	hl
78    006E  CD 00 00    	call	fseek_callee
```

The 16-bit value looks to be loaded into `hl` but then to push the 32-bit value expected by the function it's doing `push de; push hl` where `de` is not initialized.

The correct output should be "good-bye cruel world" in the second string but the sccz80 compile is appending instead of replacing the string stored in the memory file.

aralbrec, 2017-11-03:
[backgammon.c](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/EXAMPLES/backgammon.c)

```
zcc +zx -vn -SO3 -startup=4 -clib=new  backgammon.c -o backg -create-app

sccz80:"backgammon.c" L:179 Error:Symbol die1 is already defined
sccz80:"backgammon.c" L:179 Error:Symbol die2 is already defined
sccz80:"backgammon.c" L:186 Error:Symbol i is already defined
sccz80:"backgammon.c" L:186 Error:Symbol j is already defined
sccz80:"backgammon.c" L:186 Error:Symbol l is already defined
sccz80:"backgammon.c" L:186 Error:Symbol m is already defined
sccz80:"backgammon.c" L:186 Error:Symbol imoves is already defined
sccz80:"backgammon.c" L:186 Error:Symbol count is already defined
sccz80:"backgammon.c" L:227 Error:Symbol i is already defined
sccz80:"backgammon.c" L:227 Error:Symbol j is already defined
```

There is no sccz80 compile line in this program so I'm not sure if there are 2d arrays in it but at the moment it's failing on something very unusual:

```
void roll(void)
{
	extern int die1, die2;
	die1 = (rand() >> 8) % 6 + 1;
	die2 = (rand() >> 8) % 6 + 1;
}
```

That's `extern` inside the function and it refers to global variables declared outside the function.

suborb, 2017-11-03:
Thanks for these. I'll get them fixed up over the weekend.

Re open_memstream.c The warning is checking the wrong way round. The missing ld de, is related to the sdcc char work I was doing.

Re backgammon.c it looks like sccz80 doesn't handle repeat extern definitions of the same thing. 

I've not looked at sudoku yet, but I can see I just need to add an extra check to the type matcher for array/pointers.

aralbrec, 2017-11-04:
For sudoku, it looks like sccz80 is missing a division:

```
_divu_(&coord, (uint16_t)(sc - puzzle.grid), 9);
```

translates to:

```
641   00AF  21 04 00    	ld	hl,4	;const
642   00B2  39          	add	hl,sp
643   00B3  E5          	push	hl
644   00B4  21 11 00    	ld	hl,17	;const
645   00B7  39          	add	hl,sp
646   00B8  5E          	ld	e,(hl)
647   00B9  23          	inc	hl
648   00BA  56          	ld	d,(hl)
649   00BB  D5          	push	de
650   00BC  21 00 00    	ld	hl,_puzzle
651   00BF  44          	ld	b,h
652   00C0  4D          	ld	c,l
653   00C1  09          	add	hl,bc
654   00C2  09          	add	hl,bc
655   00C3  29          	add	hl,hl
656   00C4  D1          	pop	de
657   00C5  EB          	ex	de,hl
658   00C6  A7          	and	a
659   00C7  ED 52       	sbc	hl,de
660   00C9  E5          	push	hl
661   00CA  21 09 00    	ld	hl,9	;const
662   00CD  E5          	push	hl
663   00CE  CD 00 00    	call	_divu__callee
```

You can see the subtraction for `sc - puzzle.grid` but this is a subtraction of two `struct sudoku_cell *` which means the result of the subtraction should be divided by `sizeof(struct sudoku_cell *)` which is 6.  This failure to divide may be because of the mismatching pointers error.

sdcc is generating the correct code:

```
1361  0097  DD 7E 04    	ld	a,(ix+4)
1362  009A  D6 00       	sub	a, +((_puzzle) & 0xFF)
1363  009C  4F          	ld	c, a
1364  009D  DD 7E 05    	ld	a,(ix+5)
1365  00A0  DE 00       	sbc	a, +((_puzzle) / 256)
1366  00A2  47          	ld	b, a
1367  00A3  21 06 00    	ld	hl,0x0006
1368  00A6  E5          	push	hl
1369  00A7  C5          	push	bc
1370  00A8  CD 00 00    	call	__divsint_callee
```

suborb, 2017-11-04:
Yeah, that's why the warning came up in that compile - arrays weren't being treated as pointers at the appropriate point in time. It looks like the old code was multiplying by 6 before doing the subtraction which is "interesting". The new code is as follows.

I need to write some testcases and then I'll push this evening.

```
        ld      hl,17   ;const
        add     hl,sp
        ld      e,(hl)
        inc     hl
        ld      d,(hl)
        ld      hl,_puzzle
        ex      de,hl
        and     a
        sbc     hl,de
        ld      de,6
        ex      de,hl
        call    l_div_u
```

aralbrec, 2017-11-08:
A few example programs are still not functioning correctly.

sudoku from above does now compile without warnings but it isn't running correctly yet.

The same example (remove spaces)

```
790 000 300
000 006 900
800 030 076
000 005 002
005 418 700
400 700 000
610 090 008
002 300 000
009 000 054
```

produces incorrect results.  A zsdcc compile works properly.  I'm pretty sure it used to work although I did not use this particular example with the previous version of sccz80.

One of the new example programs is [dzx7](https://github.com/z88dk/z88dk/tree/master/libsrc/_DEVELOPMENT/EXAMPLES/zxn/dot-command/dzx7), a file decompressor in dot command form for the spectrum with esxdos or the next.  This one's more difficult to emulate but I am using zesarux atm.

The help text is properly printed if ".dzx7" is entered on its own.  However if you try to decompress something as in ".dzx7 thing.zx7 thing.txt" then a bunch of garbage is printed at the bottom of the screen.  What should be happening is a "M RAMTOP no good" error because ramtop needs to be below 32767 so that there is space for the decompression buffer.  The zsdcc compile works fine.

I will come back to these later on.

suborb, 2017-11-08:
I've reverted back to the copt rules from the 10/9/2017 (I just happened to have the kit lying around) and it now works correctly with the new sccz80 (as well as an old one).

So, I'll be spending the rest of my evening investigating what rule changes broke.

suborb, 2017-11-08:
Sometimes you just get lucky...

aralbrec, 2017-11-09:
Yes it looks like sudoku is solved.

dzx7 is stdarg again.

```
int error(char *fmt, ...)
{
   unsigned char *p;
   static unsigned char ebuf[65];
   
   va_list v;
   va_start(v, fmt);

#ifdef __SCCZ80
   vsnprintf(ebuf, sizeof(ebuf), va_ptr(v,char *), v);
#else
   vsnprintf(ebuf, sizeof(ebuf), fmt, v);
#endif

   for (p = ebuf; p = strchr(p, '\n'); )
      *p = '\r';
   
   ebuf[strlen(ebuf)-1] += 0x80;  
   return (int)ebuf;
}
```

Was it changed recently for the l->r case?  As things stand it looks like the argument pointer set by va_arg is pointing at the MSB of `fmt` on the stack instead of the first byte of the argument following `fmt`.  There also seems to be some confusion about where the argument pointer should point to to gather parameters.  In the classic printf implementation, it looks like the argument pointer points at the second byte of the next argument so that gathering an int is done by "ld e,(hl); inc hl; ld d,(hl); dec hl; dec hl" where the last two decrements go to the next arg.  In newlib the argument pointer is pointing at the first byte of the argument so that gathering an int is done by "ld d,(hl); dec hl; ld e,(hl); dec hl".

If that's how classic is doing it, I can make different macros for newlib and classic in stdarg.h.

suborb, 2017-11-09:
I don't think anything has changed there, obviously it has though.

I'm tempted just to switch vaarg to r->l so we just have a single macro for both compilers.

aralbrec, 2017-11-13:
Well it would solve a lot of problems but I'm not sure you can get away with it?  Will someone cast a regular function as var arg and pass as a function pointer parameter?  Maybe that's too niche to worry about.

suborb, 2017-11-13:
I'm not sure either to be honest, which is why I haven't done it. 

That case is quite niche, but it does exist for the void (*func)() case. However (I need to check), sccz80 does copy across the __z88dk_callee/__smallc state of the RHS onto the lvalue so I think it would sort itself out if the assignment and usage are close.

However, if a func pointer was passed as a parameter to a function. In which case the status is lost, but there is a warning at the call stage if the types are mismatched. That's getting even more obscure though.

The other thing to do is to switch over to r2l completely, and take some time to sort qsort, bsearch etc The rest of the library functions can stay as l2r until we get around to updating them.

Fabrizio-Caruso, 2017-12-08:
The warning "unterminated // comment" is annoying and useless for me.

I do use comments with no content in some cases and it does not seem to be worth producing a warning.

suborb, 2017-12-08:
@Fabrizio-Caruso that's the new preprocessor. I'm glad it's not caused any other issues. I'll remove the warning.

suborb, 2017-12-08:
@Fabrizio-Caruso it took me a while to reproduce it, but looking at your project I found some. It's because you've got a missing newline at the end of the file (display_macros.h, sound_macros.h).

The new preprocessor is C99 compliant which is why it's picking it up. The relevant section is C99 5.1.1.2/1: A source file that is not empty shall end in a new-line character, which shall not be immediately preceded by a backslash character before any such splicing takes place.

closed 2018-01-28 by suborb