z88dk/z88dk #491: sccz80: problems with enum

aralbrec, 2017-11-03:
zzz.c
```
#include <stdint.h>

typedef enum eNoteEdit
{
   EDIT_NONE,
   EDIT_START_TIME,
   EDIT_POS,
   EDIT_DURATION,
} ENoteEdit;

ENoteEdit g_noteEditMode = EDIT_NONE;

typedef struct
{
   uint8_t     note;    // 0-127
   uint8_t     dur;     // 0-32
   uint8_t     time;    // 0-32
   uint8_t     vel;     // 0-15
} EditDataNoteStep;

typedef struct
{
   uint8_t        index;
   EditDataNoteStep  notestep;
} SelectedEditDataNoteStep;

extern SelectedEditDataNoteStep* editedNoteStep;

void main(void)
{
   EditDataNoteStep* editDataNoteStep;
   
   g_noteEditMode = EDIT_DURATION;
   editDataNoteStep = &editedNoteStep->notestep;
}
```

zcc +zx -vn -a -clib=new zzz.c

```
sccz80:"zzz.c" L:33 Error:Must be lvalue
sccz80:"zzz.c" L:33 Warning:Expected ';'
sccz80:"zzz.c" L:34 Warning:Assigning 'editDataNoteStep', type: struct 0__anonst
ruct_0 EditDataNoteStep*editDataNoteStep from struct 0__anonstruct_0 notestep
```

suborb, 2017-11-03:
Thanks again, I understand this one too, instantiated enums are marked as KIND_ENUM which is wrong.

Incidentally, I was trying to compile the newlib sms examples and appmake failed on all of them. (Getting to the point), CandyKidDemo redefines enum_direction_type, which results in a sccz80 and clang error, but sdcc accepts.

aralbrec, 2017-11-04:
Not all of them were compilable with sccz80 before.  I just tried the moggy one and it does compile but the game doesn't work correctly.  See what happens after the identified problems are fixed.

suborb, 2017-11-04:
I can't get any newlib SMS examples to create a .sms file - it looks like appmake just exits.

However, I've looked at the asm for moggy and it's now functionally the same as before I broke everything.

closed 2017-11-04 by suborb

aralbrec, 2017-11-05:
>I can't get any newlib SMS examples to create a .sms file - it looks like appmake just exits.

Which ones?  They are all working for me.

Moggy looks ok, as do the other new sms examples I tried that will compile.   Baluba Balok now compiles with sccz80 and I don't think it did before.  I tried [AstroForce](https://github.com/z88dk/z88dk/tree/master/libsrc/_DEVELOPMENT/EXAMPLES/sms/AstroForce) too, I'm not sure if it will succeed, but the attempt did uncover an error - sccz80 crashes on windows at least.

```
zcc +sms -v -c -clib=new --constsegBANK_02 src/bank2.c -o ./bank2.o
zcc +sms -v -c -clib=new --constsegBANK_03 src/bank3.c -o ./bank3.o
zcc +sms -v -c -clib=new --constsegBANK_04 src/bank4.c -o ./bank4.o
zcc +sms -v -c -clib=new --constsegBANK_05 src/bank5.c -o ./bank5.o
zcc +sms -v -c -clib=new --constsegBANK_06 src/bank6.c -o ./bank6.o
zcc +sms -v -c -clib=new --constsegBANK_07 src/bank7.c -o ./bank7.o
zcc +sms -v -c -clib=new --constsegBANK_08 src/bank8.c -o ./bank8.o
zcc +sms -v -c -clib=new --constsegBANK_09 src/bank9.c -o ./bank9.o
zcc +sms -v -c -clib=new --constsegBANK_0A src/bank10.c -o ./bank10.o
zcc +sms -v -c -clib=new --constsegBANK_0B src/bank11.c -o ./bank11.o
zcc +sms -v -c -clib=new --constsegBANK_0C src/bank12.c -o ./bank12.o
zcc +sms -v -c -clib=new --constsegBANK_0D src/bank13.c -o ./bank13.o
zcc +sms -v -c -clib=new --constsegBANK_0E src/bank14.c -o ./bank14.o
zcc +sms -v -c -clib=new --constsegBANK_0F src/fixedbank.c -o ./fixedbank.o
zcc +sms -v -startup=16 -clib=new @zproject.lst -o astroforce -pragma-include:zpragma.inc -create-app
```