z88dk/z88dk #497: (sccz80) crash when compiling astroforce

suborb, 2017-11-05:
I don't think it previously compiled so no regression par se

```
zcc +sms -v -startup=16 -clib=new @zproject.lst -o astroforce -pragma-include:zpragma.inc -create-app
```

Narrowed down to this snippet:

```
typedef void (*MyInitEnemyFunction)( void *);

extern MyInitEnemyFunction updateintro3objectfunctions[];

int func()
{
        int val;

        (*(updateintro3objectfunctions[val]))(0);
}
```

Caused by the deref of the function pointer making a function type and calling callfunction() as if it's a function rather than a function pointer. 

I remember writing a test for this style of usage, but I seem not to have committed it.

suborb, 2017-11-05:
Noticed a problem with extern [] arrays as well. This was address as well. Astroforce now compiles with sccz80. With the old decl parser it never got close to compiling.

closed 2017-11-05 by suborb

aralbrec, 2017-11-05:
Very nice; that's definitely a step forward.  It's certainly more pleasant than waiting 20 minutes to finish compiling :)

The sccz80 compile is quite a bit larger though and too big to fit so the sms file will not run properly (appmake says it goes over by 5400 bytes, with -O3 it's about 4700).  We should probably have a look at a later date to see if anything can be done but I suspect it's down to how this program is written with high dependence on local variables.  A plain sdcc compile barely fits and a zsdcc compile has 1900 bytes free when optimization is way up.

The one thing that caught my eye is the different sizes of BANK_0F:

sccz80:  `Adding bank 0x0F (2349 bytes free)`
zsdcc: `Adding bank 0x0F (1685 bytes free)`

BANK_0F comes from [fixedbank.c](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/EXAMPLES/sms/AstroForce/src/fixedbank.c) : 
zcc +sms -v -c -clib=sdcc_iy --constsegBANK_0F src/fixedbank.c -o ./fixedbank.o

This is all data and both sccz80 and zsdcc should generate exactly the same bank contents.  I am now thinking that sccz80 is putting some of that in the wrong section and that is the reason for the large 5k code size difference.

I will check on this later if you don't look at it first.

suborb, 2017-11-05:
Nice spot, it looks like const arrays of function pointers aren't going into rodata.