z88dk/z88dk #504: [NASCOM][conio] ansi/conio gotoxy non-working in Nascom target

Fabrizio-Caruso, 2017-11-08:
It seems that gotoxy from conio is ignored by the NASCOM target.

If I compile my game CROSS CHASE on this target I see a sequence of characters being displayed from left to write.

zx70, 2017-11-15:
This is odd, I tried ansitest on it and it seems to work correctly.
Could it be that your games exceeds the NASCOM screen resolution ?
It is a small 48x16 text windows.

Fabrizio-Caruso, 2017-11-16:
My game should work on any text resolution (big enough to contain about 20 characters). It does work fine with 24x16 and 20x19 resolutions and may even work with a resolution as low as 16x8.

zx70, 2017-11-16:
I prepared examples/console/coniotest.c
It works on the ZX Spectrum (different resolutions) and Galaksija,  I already noticed it fails on the ZX81 in text mode but we now have a small test program to understand what's happening.

zx70, 2017-11-24:
Fabrizio, could you run coniotest.c on the NASCOM ?

Fabrizio-Caruso, 2017-12-10:
I can confirm the problems with the 2017/12/10 snapshot and my game BUT gotoxy worked once.
kbhit on the other hand never worked.
I know it seems weird but the compiler is not behaving deterministically.
I am now not able to get gotoxy to work with my game.
It looks like some memory leak is producing indeterministic behaviour.

Fabrizio-Caruso, 2017-12-13:
Loading a compiled version of coniotest with Virtual Nascom produces a sort of reboot with the system asking for the number of pages and then a BASIC prompt.

Fabrizio-Caruso, 2017-12-14:
The problem with the display was caused by a wrong compile line.
There are still problems with the keyboard input:
- kbhit() blocks the computer (see example below)
- getk() does not detect more than one key press when a key is kept pressed
``
		#include<conio.h>
		void WAIT_PRESS(void)
		{
			while(kbhit())
				cgetc();
			while(!kbhit())
			{ 
			}; 
			cgetc();
		}
``

zx70, 2017-12-15:
I *could* have fixed it now, please try after the next valid build.

zx70, 2017-12-24:
not fixed yet,  the  keyboard scanning code needs to be rewritten,  probably

zx70, 2018-02-08:
to be continued in #590

closed 2018-02-08 by zx70