z88dk/z88dk #519: (sccz80) Upgrade preprocessor to support variadic macros

suborb, 2017-11-15:
The cpp preprocessor doesn't support variadic macros and probably a load of other c99 features that will trip it up.

Rather than upgrading zcpp it might be best to switch to a different implementation (e.g. ucpp) which we can then use. 

It would be good for it to work with sdcc sources as well, so we only have to support a single pre-processor and hence we can get #asm into zpragma and transform it to the sdcc equivalents.

suborb, 2017-11-18:
@aralbrec didn't you have some source that failed with zcpp but worked with zsdcpp? Can you point me in the right direction?

aralbrec, 2017-11-19:
Yes but I can't remember where it was.

There is a difference in [intrinsic.h](https://github.com/z88dk/z88dk/blob/master/include/_DEVELOPMENT/sdcc/intrinsic.h):

zsdcpp
```
extern void intrinsic_ldi(void*,void*) __z88dk_callee;
#define intrinsic_ldi_helper(dst,src,num)  { intrinsic_ldi(dst,src); intrinsic_ldi_num_##num(); }
#define intrinsic_ldi(dst,src,num) intrinsic_ldi_helper(dst,src,num)
```

zcpp
```
extern void intrinsic_ldi(void*,void*) __z88dk_callee;
#define intrinsic_ldi(dst,src,num) { intrinsic_ldi(dst,src); intrinsic_ldi_num_##num(); }
```

I had to insert an intermediate step for zsdcpp to expand the `intrinsic_ldi` macro correctly.

I also had trouble getting the interrupt service routine macros working in [im2.h](https://github.com/z88dk/z88dk/blob/master/include/_DEVELOPMENT/sdcc/im2.h) but I don't see any difference in the preprocessors there, only a difference in which inlined asm construct is used.

> It would be good for it to work with sdcc sources as well, so we only have to support a single pre-processor and hence we can get #asm into zpragma and transform it to the sdcc equivalents.

sdcc also defines a bunch of its own pragmas.  I don't see how zsdcpp would communicate them to sdcc so maybe sdcc itself parses pragmas in the c source.  If the preprocessor is to be replaced, that would have to be checked out.

suborb, 2017-11-27:
I've applied the change to sccz80 only.

closed 2017-11-27 by suborb