z88dk/z88dk #524: Win32 'nightly' zip file incomplete ?

zx70, 2017-11-17:
--- Building ZX81 Library ---

C:/MinGW/bin/make zx81deps
make[1]: Entering directory 'c:/z88dk/libsrc'
make[1]: Leaving directory 'c:/z88dk/libsrc'
C:/MinGW/bin/make -C ctype
make[1]: Entering directory 'c:/z88dk/libsrc/ctype'
obj/z80/x/c:: Invalid argument
[WARN ] (lib/alloc.c:122 errno: Invalid argument) memory leak (39 bytes) allocated at z80asm.c:254
[WARN ] (lib/alloc.c:122) memory leak (12 bytes) allocated at z80asm.c:281
[WARN ] (lib/alloc.c:122) memory leak (8 bytes) allocated at z80asm.c:276
make[1]: *** [Makefile:22: obj/newlib-z80] Error 1
make[1]: Leaving directory 'c:/z88dk/libsrc/ctype'
make: *** [Makefile:373: zx81_clib.lib] Error 2
bash-3.1$

.. ctype is an almost empty folder

suborb, 2017-11-17:
ctype should only have a Makefile and a lst file in it. The actual code is pulled in from newlib.

I recently changed the build to use z80asm globbing and output directory - have you got a rebuilt z80asm?

This line `obj/z80/x/c:: Invalid argument` makes it look like the globbing isn't working in properly and is scanning directories as well as files. Maybe one for @pauloscustodio

pauloscustodio, 2017-11-17:
I will have a look

zx70, 2017-11-17:
My dumb attempt to rebuild z80asm with mingw did not help

pauloscustodio, 2017-11-17:
The -O option is concatenating the input path to the target directory, resulting in obj/z80/x/c:/...
I will remove the colon in this case.

pauloscustodio, 2017-11-17:
I've changed z80asm to accept a -O=dir and an absolute path. The mingw build passed the "make -C ctype" step but now fails further on:
```
make -C adt-newlib
make[1]: Entering directory `/c/Users/paulo/git/z88dk/libsrc/adt-newlib'
make[1]: execvp: z80asm: Bad file number
make[1]: *** [obj/newlib-z80] Error 127
make[1]: Leaving directory `/c/Users/paulo/git/z88dk/libsrc/adt-newlib'
make: *** [z88_clib.lib] Error 2
make: Leaving directory `/c/Users/paulo/git/z88dk/libsrc'
```

The makefile is passing the list of asm files in the command line (56450 characters in my system), and I suspect it is too large for windows.

suborb, 2017-11-17:
Okay. I guess we need to pass through a glob expression and get z80asm to do the expansion. I’m should be able to that change tomorrow.

suborb, 2017-11-18:
I'm currently on my third iteration of not creating a long command line, but then realising I'm creating a long command line.

As a result, I might drop the dependency check since assembling is fairly quick and stat()ing a 1000 or so files probably isn't that speedy on some file systems.

suborb, 2017-11-18:
@zx70 - can you confirm that it now works?

pauloscustodio, 2017-11-19:
> As a result, I might drop the dependency check since assembling is fairly quick and stat()ing a 1000 or so
> files probably isn't that speedy on some file systems.

If you pass -d to the assembler, it will do the stat()ing and assemble only if out of date.

suborb, 2017-11-19:
Does that work with the output directory option?

pauloscustodio, 2017-11-19:
> Does that work with the output directory option?
Yes, it compares the date of the source file against the object file in the -O directory. It cannot know if the object files were built with different -D and --cpu options, but as long as these have different output directories, it should work fine.

zx70, 2017-11-19:
I should be able to test something tomorrow

zx70, 2017-11-19:
Tested right now,  it works,  thanks for the impressive teamwork  :)

closed 2017-11-19 by zx70