z88dk/z88dk #526: Warnings on sccz80

zx70, 2017-11-20:
I'm opening this thread to help identifying few examples where the sccz80 warnings look inaccurate.
I want to point out that I perfectly see the benefit of this extra feature the way it works now, but it is still a bit annoying.

sccz80:"regexp.c" L:755 Warning:Control reaches end of non-void function

libsrc/regex(
	if (regmatch(prog->program + 1)) {
		(...)
		return(1);
	} else
		return(0);
}

zx70, 2017-11-20:
c128/psg_volume.c
----------------

sccz80:"c128/psg_volume.c" L:17 Warning:Operation on different signedness!
sccz80:"c128/psg_volume.c" L:18 Warning:Operation on different signedness!

void psg_volume(unsigned char channel, unsigned char volume) {
	volumesid(volume*15/10,0);
}

..where volumesid expects "ushort", which should be mapped to "unsigned int".   Can the unused "channel" parameter part of the problem ?

zx70, 2017-11-21:
The "end of non-void function" warning is probably an opportunity to slightly optimize the code making a default return value explicit, perhaps it can stay the way it is  ?

Fabrizio-Caruso, 2017-11-22:
I get tons of 
"Warning:Operation on different signedness!"
on things like:
extern unsigned short ghostLevel;
extern unsigned char level;
...
unsigned short computeGhostSlowDown(void)
{
        if(ghostLevel<1000)
                return INITIAL_ENEMY_SLOWDOWN-level*200-ghostLevel*16;
        return 1000;
}

It may be a good warning but it should be possible to silence it. How?

suborb, 2017-11-22:
@Fabrizio-Caruso I fixed that one in 5e8c191

zx70, 2017-11-24:
The compilation experience looks much better now, I'm closing the issue.

closed 2017-11-24 by zx70