z88dk/z88dk #528: sccz80 - asm label names reused on othello.c

zx70, 2017-11-22:
Easy to reproduce, just attempt to compile othello.c for your preferred target.

(..)
Error at file 'othello.c' line 4062: symbol 'i_0' already defined
1 errors occurred during assembly
Errors in source file othello.c:
Error at file 'othello.c' line 4062: symbol 'i_0' already defined

suborb, 2017-11-22:
I can't reproduce this on master, and neither can Travis.

So windows/environment issue?

aralbrec, 2017-11-22:
I'm getting it too, something is wrong there for windows.

Here's the generated asm for the problem area:

```
.i_6619244
	ld	hl,-53240	;const
	add	hl,sp
	ld	sp,hl
	jp	i_0
```

The `i_6619244` label is bizarre, the `-53240` constant looks wrong and `i_0` is not defined anywhere.

The error:

```
Error at file '../othello.c' line 4166: symbol 'i_6619244' already defined
                   ^ ---- (null)Warning at file '../othello.c' line 1436: intege
r '-53240' out of range
                   ^ ---- (null)Error at file '../othello.c' line 1439: symbol '
i_0' not defined
                   ^ ---- (null)Warning at file '../othello.c' line 4167: intege
r '-53240' out of range
                   ^ ---- (null)Error at file '../othello.c' line 4170: symbol '
i_0' not defined
                   ^ ---- (null)
```

aralbrec, 2017-11-22:
That bit of code is orphaned - nothing refers to it.  It's emitted just before main at the end of the game() function:

```
	C_LINE	888,"../othello.c"
	jp	i_194
.i_195
;}
	C_LINE	889,"../othello.c"
	inc	sp
	pop	bc
	pop	bc
	pop	bc
	ret

.i_6619244
	ld	hl,-53256	;const
	add	hl,sp
	ld	sp,hl
	jp	i_0
;int main()
	C_LINE	892,"../othello.c"
;{
	C_LINE	893,"../othello.c"

; Function main flags 0x00000000 __stdc 
; int main()
._main
	C_LINE	893,"../othello.c"
```

suborb, 2017-11-22:
For reference, here's the output on macOS (64 bit and 32bit) at that point:

```
        C_LINE  888,"othello.c"
        jp      i_188
.i_189
;}
        C_LINE  889,"othello.c"
        inc     sp
        pop     bc
        pop     bc
        pop     bc
        ret

;int main()
        C_LINE  892,"othello.c"
```

So something has gone wrong earlier on and the label offsets are out. My best guess would be goto handling - that may be goto cleanup code triggering for some reason.

suborb, 2017-11-22:
Can someone say when it broke? Some diagnostics for goto handling can be got out with the option -debug=99

aralbrec, 2017-11-23:
The debug=99 output:

```
C:\Projects\z80\z88dk\z88dk\examples\spectrum>zcc +zx -vn -lndos -O3 -create-app
 -zorg=50000 -DGRAPHICS ../othello.c -I.. -oothello.bin -debug=99
sccz80:"../othello.c" L:465 Debug:#11: Caughtarg!! %c
sccz80:"../othello.c" L:603 Debug:#11: Caughtarg!! %u-%u
sccz80:"../othello.c" L:619 Debug:#11: Caughtarg!! Move:
sccz80:"../othello.c" L:621 Debug:#11: Caughtarg!! Move?
sccz80:"../othello.c" L:622 Debug:#14: goto is -->flush<--

sccz80:"../othello.c" L:622 Debug:#14: Adding symbol to table

sccz80:"../othello.c" L:622 Debug:#14: Adding goto label: 00goto_getmov_flush

sccz80:"../othello.c" L:628 Debug:#14: goto is -->flush<--

sccz80:"../othello.c" L:628 Debug:#14: Sym found but still on goto

sccz80:"../othello.c" L:628 Debug:#14: Adding goto label: 00goto_getmov_flush

sccz80:"../othello.c" L:633 Debug:#14: goto is -->flush<--

sccz80:"../othello.c" L:633 Debug:#14: Sym found but still on goto

sccz80:"../othello.c" L:633 Debug:#14: Adding goto label: 00goto_getmov_flush

sccz80:"../othello.c" L:637 Debug:#14: goto is -->flush<--

sccz80:"../othello.c" L:637 Debug:#14: Sym found but still on goto

sccz80:"../othello.c" L:637 Debug:#14: Adding goto label: 00goto_getmov_flush

sccz80:"../othello.c" L:640 Debug:#14: goto is -->flush<--

sccz80:"../othello.c" L:640 Debug:#14: Sym found but still on goto

sccz80:"../othello.c" L:640 Debug:#14: Adding goto label: 00goto_getmov_flush

sccz80:"../othello.c" L:643 Debug:#14: Starting chase flush

sccz80:"../othello.c" L:643 Debug:#14: Chasing 00goto_getmov_flush # 0

sccz80:"../othello.c" L:643 Debug:#14: Chasing 00goto_getmov_flush # 1

sccz80:"../othello.c" L:643 Debug:#14: Matched #1

sccz80:"../othello.c" L:643 Debug:#14: Adding label not called flush

sccz80:"../othello.c" L:647 Debug:#11: Caughtarg!!   Huh??
sccz80:"../othello.c" L:648 Warning:Control reaches end of non-void function
sccz80:"../othello.c" L:648 Debug:#14: Cleaning k♂8☺0
8☺┼♫ #0

sccz80:"../othello.c" L:654 Debug:#11: Caughtarg!! %s
sccz80:"../othello.c" L:758 Debug:#21: prev=(null) name=(null) flags 0 oflags 0
sccz80:"../othello.c" L:759 Debug:#21: prev=(null) name=(null) flags 0 oflags 0
sccz80:"../othello.c" L:760 Debug:#21: prev=(null) name=(null) flags 0 oflags 0
sccz80:"../othello.c" L:761 Debug:#21: prev=(null) name=(null) flags 0 oflags 0
sccz80:"../othello.c" L:775 Debug:#21: prev=(null) name=(null) flags 0 oflags 0
sccz80:"../othello.c" L:775 Debug:#21: prev=(null) name=(null) flags 0 oflags 0
sccz80:"../othello.c" L:775 Debug:#21: prev=(null) name=(null) flags 0 oflags 0
sccz80:"../othello.c" L:782 Debug:#21: prev=(null) name=(null) flags 0 oflags 0
sccz80:"../othello.c" L:782 Debug:#21: prev=(null) name=(null) flags 0 oflags 0
sccz80:"../othello.c" L:783 Debug:#21: prev=(null) name=(null) flags 0 oflags 0
sccz80:"../othello.c" L:783 Debug:#21: prev=(null) name=(null) flags 0 oflags 0
sccz80:"../othello.c" L:784 Debug:#21: prev=(null) name=(null) flags 0 oflags 0
sccz80:"../othello.c" L:784 Debug:#21: prev=(null) name=(null) flags 0 oflags 0
sccz80:"../othello.c" L:785 Debug:#21: prev=(null) name=(null) flags 0 oflags 0
sccz80:"../othello.c" L:785 Debug:#21: prev=(null) name=(null) flags 0 oflags 0
[... 57 lines cut ...]
sccz80:"../othello.c" L:955 Debug:#11: Caughtarg!!

sccz80:"../othello.c" L:961 Debug:#11: Caughtarg!!

sccz80:"../othello.c" L:963 Debug:#11: Caughtarg!!  You won by %d

sccz80:"../othello.c" L:964 Debug:#11: Caughtarg!!  You lost by %d

sccz80:"../othello.c" L:965 Debug:#11: Caughtarg!!  A draw

sccz80:"../othello.c" L:966 Warning:Control reaches end of non-void function
Error at file '../othello.c' line 4166: symbol 'i_4653126' already defined
Warning at file '../othello.c' line 1436: integer '-53192' out of range
Error at file '../othello.c' line 1439: symbol 'i_0' not defined
Warning at file '../othello.c' line 4167: integer '-53192' out of range
Error at file '../othello.c' line 4170: symbol 'i_0' not defined
3 errors occurred during assembly
Errors in source file ../othello.c:
Error at file '../othello.c' line 4166: symbol 'i_4653126' already defined
                   ^ ---- (null)Warning at file '../othello.c' line 1436: integer '-53192' out of range
                   ^ ---- (null)Error at file '../othello.c' line 1439: symbol 'i_0' not defined
                   ^ ---- (null)Warning at file '../othello.c' line 4167: integer '-53192' out of range
                   ^ ---- (null)Error at file '../othello.c' line 4170: symbol 'i_0' not defined
                   ^ ---- (null)
```

suborb, 2017-11-23:
Thanks, that's confirmed it. I think the issue is that the gotoq starts populating from index 1, not index 0. At the end of the function it starts looking for dangling gotos from index 0.

So, given the memory is allocated using malloc() and not calloc() there's garbage in slot 0 and you get the junk being dumped.

suborb, 2017-11-23:
I think/hope this one is sorted now - can someone confirm?

aralbrec, 2017-11-24:
Yes, works ok!

closed 2017-11-24 by aralbrec