z88dk/z88dk #534: zsdcc produces large code even with --opt-code-size -SO3 --max-allocs-per-node200000

Fabrizio-Caruso, 2017-11-23:
I am comparing apples (zsdcc in Z88DK) and CC65 with the same code (a minimalistic version of my game).
I don't know whether 6502 binaries are usually smaller than Z80 ones.

The goal is to get a very small executable for computers with no more than 8k of available ram.

This is possibile for a Vic20 with +3k expansions (6450 bytes available for the code).
I am compiling (a minimalistic version of) my game for the Vic20+3k (total 8k) with CC65 with the highest optimization and with an option that makes all local variables static variables.
The Vic20 binary is obtained instatly. The binary is close to the limit (6427). The game works fine.

I am compiling with zsdcc with +zx --opt-code-size -SO3 --max-allocs-per-node200000, hoping to get an executable for a Spectrum 16k (which should have a bit more than 8k of available ram for the code).
The compilations takes a few minutes. The result is a binary of 9089 bytes which cannot be loaded on a Spectrum 16k... We get 50% more bytes and it takes 100 times more time.

Remark: My code is written with 8 bit computers/consoles in mind (almost only unsigned char, few local variables and parameters and lots of globals, lots of code re-used in multiple ways,  etc.)

Maybe peep hole optimization is very inefficient because it takes longer and it produces large binaries.
Maybe something else is taking up too much space. I am using startup=1 to get native directives but on the CC65 I am using conio which is probably occupying space.
Maybe we need an option to make all local variables static to use at the user's own risk: the functions are no longer reentrant but you save some space and time.

suborb, 2017-11-23:
(z)sdcc is glacial at times to be honest.

The peepholer in sdcc is context aware so it should be better than a pure blind one.

First thing to do is to run with -Ca-v -Cl-v and see which section and file is taking up most of the space - is it your code or the runtime library that's contributing the most. 

Some ideas:

For ZX/classic you'll be losing 768 bytes for the 64 column font to start off with - if you're not using that mode you could point the font to any random address to save pulling it in.

With classic/sdcc you'll also be pulling a whole bunch of printf formatters by default, so use `#pragma printf "%c %d"` to only pull in the formatters you need - replace `%c %d` with the formatters you actually use.

suborb, 2017-11-23:
Code density paper: http://web.eece.maine.edu/~vweaver/papers/iccd09/iccd09_density.pdf - it looks like z80 code is denser than 6502, though it obviously depends on the compiler.

aralbrec, 2017-11-24:
As suborb mentioned, zsdcc is known to be very slow - I have examples that take 20-40 minutes to compile on my laptop at highest optimization.  zsdcc is an optimizing compiler so it works differently than sccz80 or cc65 but still it was obviously written in a fashion that didn't consider runtime.  The peepholer it uses is not blind - it's able to follow branches, detect what registers are affected by instructions and function calls, etc.  We do a lot to clean up zsdcc's code with the peephole rules at -SO3 level but it's not complete for 32-bit or 64-bit types yet.

The first thing to do is reduce the feature set that is automatically enabled for a particular compile.  This depends on whether you're using the newlib or classic lib.  There are things to consider like turning off stdin if you're not using it, restricting the number of % converters that gets compiled with printf, disabling the heap, etc.

If you can show your compile line maybe we can make some more concrete suggestions.

Fabrizio-Caruso, 2017-11-24:
I am trying to compile CROSS CHASE
https://github.com/Fabrizio-Caruso/CROSS-CHASE/releases
for the Spectrum 16K (and other Z88DK targets with about 8k-16k of ram)
with:
spectrum_newlib_tiny:
	$(Z88DK_PATH)$(MYZ88DK) +zx --opt-code-size  -startup=1 -SO3 --max-allocs-per-node200000 -zorg=24500 -clib=sdcc_iy -DNO_SLEEP -DNO_INITIAL_SCREEN -DNO_RANDOM_LEVEL -DNO_TEXT -DTINY_GAME -vn  -D__SPECTRUM__ -create-app -o $(BUILD_PATH)/TINY_spectrum_newlib.prg $(SOURCE_PATH)/spectrum/spectrum_graphics.c $(SOURCE_PATH)/display_macros.c  $(SOURCE_PATH)/enemy.c $(SOURCE_PATH)/level.c $(SOURCE_PATH)/character.c $(SOURCE_PATH)/text.c $(SOURCE_PATH)/strategy.c $(SOURCE_PATH)/input_macros.c $(SOURCE_PATH)/main.c

aralbrec, 2017-11-24:
That's a newlib compile (`-clib=sdcc_iy`).

There are some things you can do right off the bat before taking a look at what code is being linked (suborb's other suggestion).

Create a file `zpragma.inc` that will hold options communicated to the crt startup code.  Most of the crt options for newlib are described [here](https://www.z88dk.org/wiki/doku.php?id=libnew:target_embedded#crt_configuration) and for the spectrum the default values are [here](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/target/zx/crt_config.inc#L19).  I'm looking at the spectrum defaults and seeing what can be changed to reduce program size.

`zpragma.inc`
```
// change program org
// #pragma output CRT_ORG_CODE = 24500   // an alternative to -zorg on the compile line

// move the stack to available memory
#pragma output REGISTER_SP = -1      // must be placed somewhere in the 16k ram

// halt program on exit, no need to play well with basic
#pragma output CRT_ON_EXIT = 0x10001

// eliminate the malloc heap
#pragma output CLIB_MALLOC_HEAP_SIZE = 0

// eliminate the stdio heap (no files can be opened)
#pragma output CLIB_STDIO_HEAP_SIZE = 0

// eliminate unused printf converters
# pragma printf = "%c %s"    // placed used converters here, empty string if none
```

On the 16k spectrum, the default stack location is not good (it's at address 65368).  The -1 value is going to leave it where it is, which will be set by the basic loader.  Since your program loads at 24500, the stack will be below that address.

On exit, the crt is told to halt the program.  Not having to return to basic means the crt doesn't have to take care of some housekeeping duties to allow that.

The heaps are eliminated.  If you are using malloc you should set the malloc heap size to a reasonable number.

The printf converters -- go through your code and find out what % converters you are using and list them here.  The default is to pull in most converters expected by a c program - %s, %d, %u, %ld, %lx, etc.  Notice that %u is different from %lu is different from %llu.  If you're using two 16-bit converters like %d and %u, try to use one instead if you can get away with it.

Add `-pragma-include:zpragma.inc` to your compile line to get the crt to use your pragmas.

The next step is to customize the driver instantiation.  I'm guessing you're not using scanf yet stdin will be present in the compile.  To specify what devices are instantiated on file descriptors, you can add another pragma to your `zpragma.inc` file:

```
// Use the custom driver instantiation provided in "crt_driver_instantiation.asm.m4"
#pragma output CRT_INCLUDE_DRIVER_INSTANTIATION = 1
```

Then you must create a new file with the instantiation in it:

`crt_driver_instantiation.asm.m4`
```
  ; # fd=0, stdin

   include(`../m4_file_absent.m4')dnl
   m4_file_absent

   ; # fd = 1, stdout

   include(`driver/terminal/zx_01_output_char_32_tty_z88dk.m4')dnl
   m4_zx_01_output_char_32_tty_z88dk(_stdout, CRT_OTERM_TERMINAL_FLAGS, 0, 0, CRT_OTERM_WINDOW_X, CRT_OTERM_WINDOW_WIDTH, CRT_OTERM_WINDOW_Y, CRT_OTERM_WINDOW_HEIGHT, 0, CRT_OTERM_FONT_8X8, CRT_OTERM_TEXT_COLOR, CRT_OTERM_TEXT_COLOR_MASK, CRT_OTERM_BACKGROUND_COLOR)dnl
```

What I did is copy the driver instantiation section out of the [startup=1 crt](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/target/zx/startup/zx_crt_1.asm.m4#L80) and I eliminated stdin and stderr (if you need stderr you can add that back).

Now a compile will eliminate stdin and stderr, although stderr is not really that big.

Give that a go and see how much is shaved off.

aralbrec, 2017-11-24:
I think we're doing more things than cc65 so the binary is going to hold more overhead.  For example, the output terminal is windowed and sits on a file descriptor so that it fulfills the unix io model.  It's also independent of the rom so no system software is being taken advantage of.  I think you've tried startup=30 before -- this one will be smaller than startup=1 because it uses the rom to print.  But it will be different control codes.  Let's see how you get on with the above change.

Fabrizio-Caruso, 2017-11-24:
Thanks @aralbrec, I think my code is not using that rand() call in this case. 
The code I use is selected by some flags (TINY_GAME, NO_RANDOM_LEVEL, etc...
Nevertheless there is still room for improvement and I am looking at little things that can be improved.

Fabrizio-Caruso, 2017-11-24:
By only using zpragma.inc without the customized driver instatiation, I can save about 600 bytes.

I get a tap file with 8453 bytes.

It should be small enough for a Spectrum 16k but if I try to load it, I get a frozen screen.
I can load it on a Spectrum 48k.

I wonder whether 8453 is too big. Somewhere I read that a Spectrum 16k has 8.8k of available ram for the code. What is the limit for the code generated by Z88DK?

I have used -zorg=24500 because I though it was necessary for Spectrum 16k. Is this correct?

suborb, 2017-11-24:
Top of memory on a 16k is 32768, so you'll need to org to around 24300 to get it to fit in.

Fabrizio-Caruso, 2017-11-24:
IT WORKS! Those 200 extra bytes made the difference!! THANKS GUYS for the great support!

Fabrizio-Caruso, 2017-11-24:
Can I use the very same zpragra.inc file for all other Z88DK targets? Including the stack setting?

closed 2017-11-24 by Fabrizio-Caruso

Fabrizio-Caruso, 2017-11-24:
So, from 8453  you got 7021 by a customized crt.
I have no tried that, yet but it looks a LOT. Please think of adding a general option to achieve such result. 1432  bytes is HUGE.

aralbrec, 2017-11-24:
> Can I use the very same zpragra.inc file for all other Z88DK targets? Including the stack setting?

You can use the zpragma file but the pragmas themselves are specific to newlib - classic has a different set.

One of the luxuries of starting from scratch for newlib was being able to design this sort of functionality in for all of its targets from the beginning and to expand on how much is customizable.

The classic library supports:  `#pragma printf, #pragma scanf, #pragma output CRT_ORG_CODE, #pragma output CRT_ORG_DATA, #pragma output CRT_ORG_BSS, #pragma output CRT_MODEL, #pragma output CRT_APPEND_MMAP`

Those are the ones in common with newlib.  Other pragmas depend on the target and you'd have to read the target's classic crt to see what is there.  A common one defines the heap, setting the stack pointer is present for some targets through STACKPTR instead of REGISTER_SP and special values like -1 may not be supported.

The pragma idea came after many targets in classic were written so they were added incrementally over time and the best supported targets have more, the less supported targets have less.

aralbrec, 2017-11-24:
> So, from 8453 you got 7021 by a customized crt.
> I have no tried that, yet but it looks a LOT. Please think of adding a general option to achieve such result. 1432 bytes is HUGE.

Last year I tried to add a bitmask pragma that would selectively exclude file descriptors.  So bit 0 reset means exclude stdin, etc.  That wasn't possible last year due to limitations in the tools but I think it is possible now so I will look at that again.

Fabrizio-Caruso, 2017-11-24:
How do I use the common  classic lib pragmas to save memory?
I could not find any reference of scanf in the asm crt file for the Aquarius.

Can I use the pragma scanf to disable stdint and still get conio input (getk, kbhit,...) to work? Show do use #pragma scanf OFF ?

Fabrizio-Caruso, 2017-11-25:
How do I use the printf pragma with clib and the sccz80?

#pragma printf = "%c %s %u %d"  removes all formatters if used with VG5K clib + sccz80.

What is the correct syntax in this case?

aralbrec, 2017-11-25:
Classic's stdio works differently.

In newlib, drivers are instantiated on file descriptors so if a crt enables stdin, stdout or stderr, the code will be in the output binary even if some streams are not used so providing your own list of drivers that chooses what streams are instantiated will reduce program size.

Classic instead treats fd in a switch and therefore relies on the linker to pull in relevant code.  So if you only printf and never scanf, you will not get scanf code or the underlying stdin code in.  If you do io through a file descriptor, the switch will bring in all stream code.

Long story short, if you don't use the scanf family, do not use a `pragma scanf` - the act of using it brings scanf into the compile (this applies to both classic and newlib).  Similar applies to `pragma printf`.  And with classic, getk, kbhit etc will pull in the code it needs no matter what.

> #pragma printf = "%c %s %u %d" removes all formatters if used with VG5K clib + sccz80.
> What is the correct syntax in this case?

That is correct syntax.  Maybe @suborb can ensure that the right things are in the vg5k crt - this should be in all classic crts now.

There's also another difference - the classic library differentiates between simple converters and complete ones.  There's a simple %d that does not do number formatting and a complete %0d that does it all.  The newlib only has converters that do everything.  So I think you have to put %0d in that string (and %0u %0s, etc) to get the complete converters that do formatting.  Maybe suborb can comment on that too.  Newlib will accept both forms as equivalent.

suborb, 2017-11-25:
That should work, there's no difference between sccz80/newlib and sccz80/classic in terms of determining printf/sprintf/fprintf format. In both cases, the compiler will itself determine what's needed and output to the zcc_opt.def file.

This works well when you have a single compile line, but for a makefile it's not so useful, so zpragma.inc is recommended in that case.

I checked the vg5k CRT and it does include the required file to handle the printf formatters. So, my only idea in this case is that something is wiping zcc_opt.def and then repopulating it with a printf which has no formatters defined. This can happen if a file that only does a printf("..."); is processed last of all and `-preserve` isn't supplied to zcc. This is most likely a makefile thing so it might be worth eyeballing the contents of zcc_opt.def at each stage of the compilation.

With regard to the flag handling for the format strings. In most cases the additional format specifiers (width, alignment etc) tend not to be used by programs so not linking the code that parses and applies them can result in a saving of around 100 bytes + any additional routines they pull in. 

If sccz80 is used then it automatically (as discussed above) handles enabling features as required, if sdcc is used then all printf specifiers are enabled by default. This way, things tend to just work without need for explicit tuning.

In terms of classic, once a single number formatter has been included then each additional formatter will be less than 20 bytes or so.

Fabrizio-Caruso, 2017-11-26:
Using %0u seems to enable the formatter in clib. Just a few bytes saved. Not much but better than nothing.

On the other hand these formatters including the ones whose name starts with CLIB have no effect on VG5K with CLIB:
// move the stack to available memory
#pragma output REGISTER_SP = -1      // must be placed somewhere in the 16k ram

// halt program on exit, no need to play well with basic
#pragma output CRT_ON_EXIT = 0x10001

// eliminate the malloc heap
#pragma output CLIB_MALLOC_HEAP_SIZE = 0

// eliminate the stdio heap (no files can be opened)
#pragma output CLIB_STDIO_HEAP_SIZE = 0

suborb, 2017-11-26:
Those are all newlib pragmas so they'll have no effect on the vg5k which is classic only.

In classic, it's your responsibility to setup the malloc heap, either by telling the crt0 to do it automatically (-DAMALLOC) or programatically (malliinit, sbrk etc) as a result there's no need to turn it off.

Looking at the crt0, I think you could save some space with `#pragma nostreams` which will disable stdio support. Possibly a couple of hundred bytes. Obviously you're using printf() which needs stdio support, so in this case you'd need to switch to use printk(). sprintf() should continue to work without stdio.