z88dk/z88dk #535: Problem with printf and friends on z80/rc2014

jayacotton, 2017-11-24:
The code below shows a problem with z80/rc2014.

When I look at the assembly output from the compiler it looks o.k.  I would expect it to just work, 
but when I run it, no joy.   I suspect there is a bug in printf (unlikely) or the rc2014/cpm crtc0 code.
I did a test that just does a printf("test\n"); that that works.  Also the hello world example code
works.  

Here is the expected output on a cygwin shell:
$ ./a.exe
File:testpf.c Line:6 
File:testpf.c Line:8 
On CPM I get:
G>testpf

G>
Here is the compile command.
zcc +cpm -create-app -no-cleanup --c-code-in-asm --list testpf.c -otestpf

This is the test program I am working on.

main(argc,argv)
int argc;
char *argv[];
{
char buff[500];
	sprintf(buff,"File:%s Line:%d ",__FILE__,__LINE__);
	printf("%s\n",buff);
	printf("File:%s Line:%d \n",__FILE__,__LINE__);
}

jayacotton, 2017-11-24:
I fixed my problem.  Seems that <stiod.h> MUST be included.

jayacotton, 2017-11-24:
verified that stdio.h must be included to work correctly.

closed 2017-11-24 by jayacotton

aralbrec, 2017-11-24:
Yes that's right.  stdio.h supplies the prototypes for printf, etc.

The cpm target is for cpm generally so there shouldn't be any dependency on a specific machine like the rc2014.