z88dk/z88dk #536: How to crash sccz80..

zx70, 2017-11-24:
I wasn't expecting to see it working, but this source example harms sccz80 in a very bad way:

#include <stdio.h>

#define CITY = 2;
#define WEEK = 7;

    int temperature[CITY][WEEK];

int main()
{
    for (int i = 0; i < CITY; ++i) {
        for(int j = 0; j < WEEK; ++j) {
            printf("City %d, Day %d: ", i+1, j+1);
            scanf("%d", &temperature[i][j]);
        }
    }

    printf("\nDisplaying values: \n\n");
    for (int i = 0; i < CITY; ++i) {
        for(int j = 0; j < WEEK; ++j)
        {
            printf("City %d, Day %d = %d\n", i+1, j+1, temperature[i][j]);
        }
    }
    return 0;
}

suborb, 2017-11-24:
City has a value of "=2" and week a value of "=7" which is syntactically incorrect.

zx70, 2017-11-24:
Understood, thank you.   The syntax error msg could be more friendly, though  :)

suborb, 2017-11-24:
Hmmm, maybe you're seeing different errors, I get this:

```
sccz80:"y.c" L:6 Error:Invalid expression
sccz80:"y.c" L:6 Error:Expecting constant expression
sccz80:"y.c" L:6 Error:Missing token, expecting ] got 2
sccz80:"y.c" L:6 Error:Missing token, expecting = got 2
sccz80:"y.c" L:9 Error:Invalid expression
sccz80:"y.c" L:9 Warning:Expected ';'
sccz80:"y.c" L:9 Error:Missing token, expecting ) got ;
sccz80:"y.c" L:9 Error:Unknown symbol: i
```

sdcc gives me this:

```
y.c:6: syntax error: token -> '=' ; column 17
```

Clang will tell you where the = comes from.

closed 2017-11-24 by suborb

zx70, 2017-11-25:
on my Windows env. I'm not able to see it,  it just crashes.  I'll experiment a bit

aralbrec, 2017-11-25:
It crashes here too after printing this:

```
sccz80:"zzz.c" L:6 Error:Invalid expression
sccz80:"zzz.c" L:6 Error:Expecting constant expression
sccz80:"zzz.c" L:6 Error:Missing token, expecting ] got 2
sccz80:"zzz.c" L:6 Error:Missing token, expecting = got 2
sccz80:"zzz.c" L:10 Error:Invalid expression
sccz80:"zzz.c" L:10 Warning:Expected ';'
sccz80:"zzz.c" L:10 Error:Missing token, expecting ) got ;
sccz80:"zzz.c" L:10 Error:Unknown symbol: i
```

"ptr" is a nullptr:

```
            } else {
00FB7B49  jmp         primary+51Ch (0FB7B9Ch)  
            /* assume it's a function we haven't seen yet */
            /* NB value set to 0 */
                warningfmt( "Implicit definition of function '%s' it will return an int. Prototype it explicitly if this is not what you want.", sname);
00FB7B4B  push        offset string "Implicit definition of function "... (0FBFCD0h)  
00FB7B50  call        warningfmt (0FB09B0h)  
00FB7B55  add         esp,8  
                ptr = addglb(sname, default_function(sname), 0, KIND_INT, 0, STATIK);
00FB7B58  lea         ecx,[sname]  
00FB7B5E  push        0  
00FB7B60  push        0  
00FB7B62  push        4  
00FB7B64  push        0  
00FB7B66  call        default_function (0FAF400h)  
00FB7B6B  mov         edx,eax  
00FB7B6D  lea         ecx,[sname]  
00FB7B73  call        addglb (0FBB580h)  
00FB7B78  mov         esi,eax  
00FB7B7A  add         esp,10h  
                ptr->size = 0;
00FB7B7D  mov         dword ptr [esi+49Ch],0  
                ptr->flags |= c_use_r2l_calling_convention == YES ? 0 : SMALLC;
00FB7B87  mov         eax,dword ptr [c_use_r2l_calling_convention (0FC3144h)]  
00FB7B8C  dec         eax  
00FB7B8D  neg         eax  
00FB7B8F  sbb         eax,eax  
00FB7B91  and         eax,200h  
00FB7B96  or          dword ptr [esi+4A4h],eax  
            }
        }
        lval->symbol = ptr;
00FB7B9C  mov         dword ptr [edi],esi  
        lval->ltype = ptr->ctype;
00FB7B9E  mov         eax,dword ptr [esi+88h]  
00FB7BA4  mov         dword ptr [edi+4],eax  
        lval->indirect_kind = KIND_NONE;
00FB7BA7  mov         dword ptr [edi+8],0  
        lval->val_type = ptr->type ; /* Null function, always int */
00FB7BAE  mov         eax,dword ptr [esi+84h]  
00FB7BB4  mov         dword ptr [edi+2Ch],eax  
        lval->flags = ptr->flags;
00FB7BB7  mov         eax,dword ptr [esi+4A4h]  
00FB7BBD  mov         dword ptr [edi+34h],eax  
        return (0);
00FB7BC0  xor         eax,eax  
    }
}
```
at address:

```
        lval->ltype = ptr->ctype;
00FB7B9E  mov         eax,dword ptr [esi+88h] 
```