z88dk/z88dk #540: zsdcc: update past sdcc revision 9958 causes compile errors

aralbrec, 2017-11-26:
Updating past 9958 causes compile errors in some test programs.  sdcc is passing its own regression tests so it's possible there is some interaction in z88dk that is not working well.

It's also been suggested that we integrate testing using sdcc's regression suite to help identify any code generation problems.

zx70, 2018-02-07:
Surprisingly Fabrizio Caruso was able to build his game with z88dk/zsdcc for the zx81 target!
It is a few thousand of lines long, and it shows the recent incredible improvements..  are the problems still existing?

aralbrec, 2018-02-07:
A compile for the zx81 should use "--reserve-regs-iy" to keep zsdcc away from iy.  This almost works but zsdcc will use iy anyway if a stack frame becomes large.

As for issues with updating to the current sdcc, I haven't tried to see if the problems still exist.  A couple of fixes in the peephole set have been implemented since the problems were first noted.

spth, 2018-04-04:
The issue of using iy despite --reserve-regs-iy has been fixed in upstream SDCC for a while. If you can reproduce such an issue using current SDCC, please file a bug report at Sourceforge.

Philipp

closed 2019-02-09 by suborb