z88dk/z88dk #541: z80asm:  PHASE does not like difference of address labels

aralbrec, 2017-11-28:
I'm getting some compile errors when using PHASE.

The relevant file is [here] and the problematic phase statement is:

```
PHASE __SYSVAR_TSTACK - (__allocate_pages_end - __allocate_pages_begin) + 1
```

`__SYSVAR_TSTACK` is a constant but `__allocate_pages_end` and `__allocate_pages_begin` are address labels.  Their difference should be a constant.

The assembler is complaining that the PHASE value is not constant:

```
Errors in source file c:\Projects\z80\z88dk\z88dk\lib\config\..\..\/libsrc/_DEVELOPMENT/target/zxn/zxn_crt.asm:
Error at file 'c:\Projects\z80\z88dk\z88dk\lib\config\..\..\/libsrc/_DEVELOPMENT/target/zxn/crt_alloc_esxdos_dotn.inc' line 3: symbol '__allocate_pages_end' not defined
                   ^ ---- PHASE __SYSVAR_TSTACK - (__allocate_pages_end - __allocate_pages_begin) + 1
Error at file 'c:\Projects\z80\z88dk\z88dk\lib\config\..\..\/libsrc/_DEVELOPMENT/target/zxn/crt_alloc_esxdos_dotn.inc' line 3: expected constant expression
                   ^ ---- PHASE __SYSVAR_TSTACK - (__allocate_pages_end - __allocate_pages_begin) + 1
Error at file 'c:\Projects\z80\z88dk\z88dk\lib\config\..\..\/libsrc/_DEVELOPMENT/target/zxn/crt_alloc_esxdos_dotn.inc' line 53: symbol '__deallocate_pages_end' not defined
                   ^ ---- PHASE __SYSVAR_TSTACK - (__deallocate_pages_end - __deallocate_pages_begin) + 1
Error at file 'c:\Projects\z80\z88dk\z88dk\lib\config\..\..\/libsrc/_DEVELOPMENT/target/zxn/crt_alloc_esxdos_dotn.inc' line 53: expected constant expression
                   ^ ---- PHASE __SYSVAR_TSTACK - (__deallocate_pages_end - __deallocate_pages_begin) + 1
```

You can reproduce this error by compiling [md5sum.c](https://drive.google.com/file/d/1W4QrRgUOot1hDzOGqFa7fIf5mGTbzW1y/view?usp=sharing) using:

```
zcc +zxn -v -startup=30 -clib=sdcc_iy -SO3 --max-allocs-per-node200000 --opt-code-size md5sum.c -o md5sum -subtype=dotn -create-app
```

pauloscustodio, 2017-11-28:
The reason for the error is that when the PHASE directive is encountered, the __allocate_pages_end is not yet known and the phase-ASMPC cannot be defined.
To fix this z80asm would have to make an extra pass through the code to compute label offsets.

Looking at the actual code in crt_alloc_esxdos_dotn.inc, except for the statement "ld hl,allocated_pages", the code is position-independent and does not require a PHASE/DEPHASE, and the allocated_pages is at a fixed distance from __allocate_pages_end. 

Am I missing something?

aralbrec, 2017-11-29:
I can do a work-around in this case but it would be a problem if the code was not nearly position-independent.

Here the program itself is about to be banked out of memory so this subroutine and its data has to be copied to a common temporary stack area, to which only a pointer to the top of the stack is available (`__SYSVAR_TSTACK`).  So I need to compute the start address of the routine by subtracting its size from the stack pointer constant and that requires a forward reference label.  Were the code not position-independent as here, I could count the byte size by hand or assume some size that won't be exceeded but that's not as nice as having the assembler do it :)

But if that's too hard to do then it can just be a documented limitation of the PHASE statement that forward referenced labels are not allowed.

pauloscustodio, 2017-11-29:
I will keep this open as a reminder that constant expressions should be able to be computed with forward references in the same file.