z88dk/z88dk #543: can't build campus lisp

zx70, 2017-11-30:
sorry guys,  I must be missing something but I don't seem to be able to build clisp.c and see it run correctly.    after the prompt it becomes unstable,   it can be a simple low memory problem,  can you reproduce the problem ?
eg with sccz80 any instruction but nil creates an error condition.

suborb, 2017-12-01:
I'll take a look this weekend. This example seems to be a bit of a canary - have we got an example program that generates output? I can add it to the tests to ensure it runs correctly.

zx70, 2017-12-01:
thankyou for answering  :)
it is sadly the bit I have so far, I'm taking aside sdcc which may require extra tuning basing on the specific page by Alvin (the text looks garbled,  as if the keyboard echo for the literal elements was mapped to a bad font; numbers and direct output look good), I'd focus on sccz80.
I compiled it in many ways,  e.g. with '-DTINYMEM' and -O3 to try to save as much memory as possible. 
I also tried to move zorg.
everytime the any attempt to compute something, eg:
(* 3 4)
.. answers once with:
ERROR.
ILLEGAL FUNCTION CALL: ( . )

...the next attempts will get no answer.

aralbrec, 2017-12-01:
For sdcc you have to compile clisp.c with `--fsigned-char` because by default it makes `char` unsigned but this program expects `char` to be signed.

I have a second version in the newlib examples that worked with both sccz80 and zsdcc so I will give those a try now.

aralbrec, 2017-12-01:
Same thing happens in the newlib version - sccz80 reports that illegal function error.  The zsdcc compile crashes.  Looks like a lot of debugging ahead :-(

zx70, 2017-12-02:
Well done, you're great as always !
...now I can add to clisp.c the optional turtle graphics functions I was keeping aside   :D

zx70, 2017-12-02:
..is it impacting also zsdcc?  otherwise we can close it

suborb, 2017-12-02:
I think @aralbrec suggests that it is. I've got: `Build: 3.6.5 #9897 (Mac OS X i386) May  4 2017` and it works ok. So if it's broken there's been a regression.

zx70, 2017-12-02:
I think the same works also on the late builds.
I only have problems by typing alpha characters (garbled symbols are shown) , which happened also before,  my thought was that it was consequence of a memory overflow/ overlap.

aralbrec, 2017-12-02:
The sccz80 compile works but an optimized zsdcc compile crashes.

Try `zcc +test -compiler=sdcc --reserve-regs-iy -SO3 --max-allocs-per-node200000 --opt-code-size --fsigned-char clisp.c`
(Make yourself a cup of coffee while that runs :)

It's going to be an issue in the peephole rules.  It crashes before printing anything so the initial bug is right at the beginning.

zx70, 2017-12-02:
I built a zx version, but even if faulty  (as explained)  it is able to do some simple algebric operation.

desertkun, 2023-01-20:
@suborb I think we have a regression

```
diff -w Issue_543_complement.opt.tmp2 results/Issue_543_complement.opt
12c12
< 	ld	hl,4	;const
---
> 	ld	hl,2	;const
14,16c14
< 	ld	e,(hl)
< 	inc	hl
< 	ld	d,(hl)
---
> 	call	l_glong2sp
18,19c16,17
< 	ld	d,h
< 	ld	e,l
---
> 	ld	de,28672
> 	call	l_long_and
make[1]: *** [Issue_543_complement.opt] Error 1
```

I wonder why CI doesn't catch this? It happens on my arm64 mac

suborb, 2023-01-20:
There's no agents in CI building on arm64!

suborb, 2023-01-20:
Continuing discussion on #2148

commit 0cf962e0 (2017-12-04): Fix #543: zsdcc peephole code-reduction-309f3 must be qualifed on BC not changing
commit c945dd9e (2023-01-21): Issue #2048: Add test, fix #543 as well?
closed 2023-01-20 by suborb