z88dk/z88dk #544: Intro to COPT language

feilipu, 2017-12-01:
I'm trying to write some peephole rules to implement an inline macro similar to the __BREAK macro provided in #374. But, I can't find a command or language reference. Is there one that I can read, please?

In summary, I want a C macro `call_far(addr, bankRel)` to be converted into this assembly:
```asm
ret 10
defw addr
defb bankRel
```
so that I can call this [`_call_far_rst` function](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/target/yaz180/driver/yabios/asm_common1_driver.asm#L37), or `_jp_far_rst`, using a `rst` instruction. There are a few of these functions that I need to attach to a `rst`, so if there's a general solution it would be better.

So far I have this... (but I'm guessing it is wrong).
```asm
;;
;; Inlined C macro call_far(addr, bankRel)
;;

%"[ \t]*"0call%"[ \t]*"1_CALL_FAR_HELPER(%0,%1)
=
	rst 10H
	defw %0
	defb %1

%"[ \t]*"0call%"[ \t]*"1CALL_FAR_HELPER(%0,%1)
=
	rst 10H
	defw %0
	defb %1

%"[ \t]*"0jp%"[ \t]*"1_CALL_FAR_HELPER(%0,%1)
=
	ld hl, %0
	ld e, %1
	rst 18H

%"[ \t]*"0jp%"[ \t]*"1CALL_FAR_HELPER(%0,%1)
=
	ld hl, %0
	ld e, %1
	rst 18H

%"[ \t]*"0defc%"[ \t]*"1%"[A-Za-z0-9_]*"4%"[ \t]*"2=%"[ \t]*"3_CALL_FAR_HELPER(%0,%1)
=
%4:
	rst 10H
	defw %0
	defb %1

%"[ \t]*"0defc%"[ \t]*"1%"[A-Za-z0-9_]*"4%"[ \t]*"2=%"[ \t]*"3CALL_FAR_HELPER(%0,%1)
=
%4:
	rst 10H
	defw %0
	defb %1
```

suborb, 2017-12-01:
There's a man page in doc/ so do man ./copt.man 

Getting the white space matching is usually the hardest bit :)

feilipu, 2017-12-01:
Thsnks. Weekend reading.

Now everything is working... time to make it pretty. :-)

feilipu, 2017-12-01:
I'm afraid with the need to pass parameters into the assembly code, simply copying the style of the #374 solution is too hard. I've made written some macros to solve my issue in the interim #545.

closed 2017-12-01 by feilipu

aralbrec, 2017-12-01:
I added some escape sequences to copt some time ago to help with whitespace.  Actually, maybe it was just one "\t".  You can see examples of its [use](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/sdcc_opt.1#L337) like this one:

```
%"[ \t]*"1.ds%"[ \t]+"2%0
=
%1DEFS%2%0
```

The `%"..."n` means match to a regular expression instead the quotes and assign it to slot `n`.  You're matching exact character sequences so it's all space sensitive.

There is already a RST substitution mechanism in z88dk.  For sdcc it is [here](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/sdcc_opt.9#L10) and is done with sdcc's peepholer and for sccz80 it is [here](https://github.com/z88dk/z88dk/blob/master/lib/z80rules.9#L2) and is done with copt.

The name of the called function has to begin with "__RSTXX" where "XX" is the hex restart address like 00,08,10,etc.  I think it will work for all call linkages.  I have used it for the sega master system as you can see [here](https://github.com/z88dk/z88dk/blob/master/include/_DEVELOPMENT/proto/arch/sms/SMSlib.h#L129).  In this header file, a function is substituted with a macro containing the "__RSTXX" prefix when a specific crt is in use (the crt that has this function on a restart).