z88dk/z88dk #546: CLIB ANSI sleep not sleeping...

Fabrizio-Caruso, 2017-12-08:
If I use sleep in my game (when compiled with clib=ANSI, sleep seems to have no effect at all.

suborb, 2017-12-08:
Which target?

Fabrizio-Caruso, 2017-12-08:
Sorry! I get the problem on the Spectrum. I have corrected the issue title and content.

Fabrizio-Caruso, 2017-12-11:
I am getting a similar related issue on the CPC: sleep sleeps much longer than expected.

zx70, 2017-12-12:
Problem found, the " __z88dk_fastcall" suffixes must be removed in <stdlib.h>

zx70, 2017-12-12:
Fabrizio, would you be so kind to refresh your z88dk copy at the next valid nightly build and confirm the fix ?

zx70, 2017-12-13:
I just reverted the fix for the Z88 only.   It has its own native functions written in assembly and the fastcall mode is the correct one for its function variants.

closed 2017-12-19 by zx70