z88dk/z88dk #548: [zx81/zx80/lambda][sccz80] Redefining gotoxy with zx_setcursorpos now produces an error

Fabrizio-Caruso, 2017-12-08:
I must admit I was not doing the cleanest coding when doing
                #define gotoxy(x,y) zx_setcursorpos(y-1,x)
but it worked and compiled with a reasonable warning about the redefinition.

The reason for this hack is that gotoxy does not work on a "normal" Zx80/Zx81.
The code would not even compile without this redefinition (it would complain about some ansi_attr already defined.

Now the warning has become an error and it is no longer possible to use conio-like functions.

This is a special case. Ideally it would be nice to have conio support on these machines or at least to allow this sort of hack.

suborb, 2017-12-08:
I think gotoxy is already a macro, so you might need to a #undef gotoxy prior to your definition. I suspect the old pre-processor silently swallowed a repeated definition.

zx70, 2017-12-12:
You have two more options:
- to remove the <conio.h> inclusion
- to define gotoxy(x,y) before including  <conio.h>

Fabrizio-Caruso, 2017-12-13:
Thanks #undef seems to solve the problem for me. It seems to work fine.
I need some (sort of) conio on this target for the unmodified ZX80. Please, if possible keep what is already available.

closed 2017-12-13 by Fabrizio-Caruso