z88dk/z88dk #552: sccz80: does not recognize forever loop so emits contradictory warning with sdcc

aralbrec, 2017-12-08:
In this one sdcc recognizes that the while loop is forever so warns about unreachable code for the return after the loop.

On the other hand sccz80 does not recognize the forever loop but instead warns that a return is missing because the function must return a value.

A contradiction (one doesn't want the return, the other does).  But I don't think anything can be done if sccz80 does not recognize an infinite loop.

```
// This program gives contradictory warnings with sccz80 and sdcc in z88dk 2017-12-07.
// In z88dk 2017-10-26, sccz80 don't warn if the return statement is missing.

// zcc +zx -vn -O3 -startup=31 -clib=new return_test.c -o return_test -create-app
// zcc +zx -vn -SO3 -startup=31 -clib=sdcc_iy return_test.c -o return_test -create-app

#include <input.h>

int main(void)
{
    while (1)
    {
        if (in_inkey())
        {
            in_wait_nokey();
            // Do some work
        }
    }

    // sccz80: If the return is missing, the warning "Control reaches end of non-void function" is issued.
    // sdcc: If the return is here, the warning "unreachable code" is issued.
    return 0;
}
```

suborb, 2017-12-08:
sccz80 doesn't do flow analysis, it just wants a return at the end. 80% of the time if there's not one then there is a problem.

I can silence the warning to satisfy the 20%.

aralbrec, 2017-12-08:
No I think it's better to have it.

suborb, 2017-12-08:
That’s my gut feeling as well.

closed 2017-12-09 by aralbrec