z88dk/z88dk #556: [OSCA] How to use the .exe file? Is it a bios replacement? a disk image? a raw binary?

Fabrizio-Caruso, 2017-12-13:
Hi! 
The page about the OSCA target
https://github.com/z88dk/z88dk/wiki/Platform---OSCA
does not say much about the output file that is produced by the compiter.
It only says it can be renamed to .exe.
It says that there are three ways to compile but does not mention any difference.

How can I use the .exe? If I run the OSCA emulator
http://www.basoft.co.uk/wordpress/emulation/computer-system-emulation/v6z80p/osca-emulator/

The option to load binary files is greyed out. 

Trying to start the exe in the Windows cmd line, does not work either.

Ideally I would like something that can be run in the easiest possible way with an emulator and/or real machine.

zx70, 2017-12-13:
Fabrizio,
the OSCA (v6z80p) computer boots its operating system (called FLOS) from a FAT formatted SDCARD.
The FLOS is able to load "EXE" files (not to be confused with the MSDOS or WINDOWS ones), which may be copied on the memory itself or on an image file, e.g. with WinImage.
The official code repository is here:
https://sourceforge.net/p/v6z80p/code/HEAD/tree/trunk/

closed 2017-12-13 by zx70

Fabrizio-Caruso, 2017-12-13:
Yes, my question is on how to copy to memory itself. Maybe some emulator has this feature. 
The option to load binary files is greyed out in the OSCA emulator I have mentioned above.