z88dk/z88dk #560: P6001 printf("abcd...") does not display (capital) letters

Fabrizio-Caruso, 2017-12-26:
P6001 printf("abcd...") does not display letters, not even capital letters as expected. 

Current behaviour on targets with no small letters is to display:
small letters as capital letters;
capital letters display target specific characters.

zx70, 2018-01-26:
I confirm the issue,  the result of my tests is:
 - the problem happens only in ANSi VT mode
 - in non-ansi mode lowercase and uppercase as both available
 - in ANSI VT mode uppercase text is shown correctly, wrong symbols (math operators etc.) are displayed in place of the lowercase text
 - the few external asm sources I could examine (eg the cp/m bios) do not seem to apply workarounds to make the lowercase text work, so I'm supposing the behaviour must be set up rather than managed inside the print function
 - a workaround to force uppercase text is obviously possible,  I'll probably surrender soon and will do that

zx70, 2018-01-30:
The workaround is now in place

closed 2018-01-30 by zx70