z88dk/z88dk #563: z80asm: consolidate object file does not like global defc

aralbrec, 2018-01-01:
I'll come up with a test example tomorrow, but this is what I am seeing from a consolidate object file:

                               GLOBAL _game_end
2599  09FF              EXTERN l_ret
2600  09FF              defc _game_end = l_ret

Error at file 'game_engine.c' line 2600: symbol '_game_end' not defined

I quickly tried creating a plain object file using something similar to the above but that seemed to work fine.

aralbrec, 2018-01-02:
@pauloscustodio 
I haven't been able to create the problem in a small example yet but the problem clearly exists in the large example I am trying now.  The code is private so I can't share all the source but if you need to see it I can maybe send you the complete source in private.

A consolidated object file "game.o" is being created from a list of source files:
`zcc +zx -v -c -clib=sdcc_iy %CFLAGS% --fsigned-char -o game @zproject.lst globals_06.o`

**zproject.lst**
```
btiles.asm
font.asm
globals.c
game.c
game_ay.asm
game_ay_res_fx.asm
game_ay_res_midi.asm
game_enemies.c
game_engine.asm
game_engine.c
game_player.c
game_sound.asm
game_sprite.c
game_zx.asm
game_zx.c
```

Checking what's in the resulting game.o:

`z80nm game.o`
```

File game.o at $0000: Z80RMF11
  Name: game
  Names:
    G A $0000 _btiles (section rodata_user) btiles.asm:5
    G A $0000 _game_font (section FONT) font.asm:5
    G A $0000 _control_method (section rodata_compiler) globals.c:712
    G A $0038 _player_lvl_table (section data_compiler) globals.c:801
    G A $0023 _sprite_frames (section data_compiler) globals.c:779
    G A $000E _sprite_speed (section data_compiler) globals.c:757
    G A $0000 _joynames (section data_compiler) globals.c:749
    G A $0391 _draw_count (section bss_compiler) globals.c:675
    G A $0389 _sprite_speed_alt (section bss_compiler) globals.c:673
    G A $0388 _map_paper_last (section bss_compiler) globals.c:671
    G A $0387 _map_paper_clr (section bss_compiler) globals.c:669
    G A $0386 _map_paper (section bss_compiler) globals.c:667
    G A $0385 _screen_ink (section bss_compiler) globals.c:665
    G A $0384 _screen_paper (section bss_compiler) globals.c:663
    G A $0383 _game_start_scr (section bss_compiler) globals.c:661
    G A $037C _game_respawn_tile (section bss_compiler) globals.c:659
    G A $0375 _game_respawn_index (section bss_compiler) globals.c:657
    G A $0367 _game_respawn_time (section bss_compiler) globals.c:655
    G A $0365 _game_score_top (section bss_compiler) globals.c:653
    G A $0364 _game_menu_sel (section bss_compiler) globals.c:651
    G A $0363 _game_over (section bss_compiler) globals.c:649
    G A $0362 _game_sound (section bss_compiler) globals.c:647
    G A $0361 _game_inmune (section bss_compiler) globals.c:645
    G A $0360 _game_god_mode (section bss_compiler) globals.c:643
    G A $035F _game_brick_tile (section bss_compiler) globals.c:641
    G A $035E _spr_hack (section bss_compiler) globals.c:639
    G A $035D _boss_stat (section bss_compiler) globals.c:637
    G A $035B _boss_time_fire (section bss_compiler) globals.c:635
    G A $0359 _boss_time (section bss_compiler) globals.c:633
    G A $0358 _boss_tile (section bss_compiler) globals.c:631
    G A $0357 _boss_inc (section bss_compiler) globals.c:629
    G A $0356 _boss_col (section bss_compiler) globals.c:627
    G A $0355 _boss_lin (section bss_compiler) globals.c:625
    G A $034D _bullet_vel (section bss_compiler) globals.c:623
    G A $0345 _bullet_frames (section bss_compiler) globals.c:621
    G A $033D _bullet_colint (section bss_compiler) globals.c:619
    G A $0335 _bullet_dir (section bss_compiler) globals.c:617
[... 329 lines cut ...]
200   0000              ;--------------------------------------------------------
201   0000              ; Public variables in this module
202   0000              ;--------------------------------------------------------
203   0000              	GLOBAL _enemy_hit
204   0000              	GLOBAL _enemy_turn
205   0000              	GLOBAL _boss_turn
206   0000              	GLOBAL _enemy_move
207   0000              	GLOBAL _enemy_static
208   0000              	GLOBAL _enemy_horizontal
209   0000              	GLOBAL _enemy_vertical
210   0000              	GLOBAL _enemy_walk
211   0000              	GLOBAL _enemy_avoid_fall
212   0000              	GLOBAL _enemy_avoid_dead
213   0000              	GLOBAL _enemy_respawn
214   0000              	GLOBAL _enemy_init
...
...
890   0000              	SECTION code_compiler
891   0000              ;	---------------------------------
892   0000              ; Function enemy_hit
893   0000              ; ---------------------------------
894   0000              	defc _enemy_hit	=	_sound_coin
```

So you can see the function `_enemy_hit` is exported as global and it is defined as a defc.  The global exports in lines 203-214 all appear in the z80nm of the consolidated object file except for `_enemy_hit` so when the consolidated object file is formed, z80asm is eliminated the defc definition of `_enemy_hit`.

pauloscustodio, 2018-01-02:
Let me a couple of days to have a look.

pauloscustodio, 2018-01-03:
I managed to reproduce the error.

commit 66879920 (2018-01-03): Fix #563 consolidated object file erroneously eliminates globally exported defc

Only external or address global symbols where being passed to the consolidated object file, leaving out DEFC.
Removed the restriction, passing all symbols to the consolidated object file.
closed 2018-01-03 by pauloscustodio