z88dk/z88dk #564: Pointer to Struct conversions not working

Fabrizio-Caruso, 2018-01-01:
While trying to do a little OOP in ANSI C through pointer to struct does not work.

Something like:
struct ParentStruct
{
  ...
};
typedef ParentStruct Parent;

struct ChildStruct
{
   Parent _parent;
... 
};
typedef ChildStruct Child;

Convert a poinyer to Child to a pointer to Parent, won't work.
It does work on other ANSI compilers, ie., you get erros anso Unknown member when trying to access
Parent's members.

This feature is very important because it allows OOP in ANSI C.

aralbrec, 2018-01-02:
I think you're just missing `struct`s in your typedefs.  C doesn't allow structs to be referred to using their names only (that's a C++ thing).

```
struct ParentStruct
{
   int a;
};
typedef struct ParentStruct Parent;

struct ChildStruct
{
   Parent _parent;
	int b;
};
typedef struct ChildStruct Child;

Child c;

void main(void)
{
	Parent *p = (Parent *)(&c._parent);
	p->a = 6;
}
```

Fabrizio-Caruso, 2018-01-03:
@aralbrec 
The missing struct was a typo (now edited). Sorry.

The problem is that sub-type polymorphism should work in ANSI C:
   Parent *p = (Parent *) &c

No need to access the c._parent explicitly.

When I use this, it does not always work as expected.

suborb, 2018-01-03:
> The problem is that sub-type polymorphism should work in ANSI C:
> When I use this, it does not always work as expected.

Have you got an example?

Fabrizio-Caruso, 2018-01-03:
I am getting an error if I cast to the "parent" struct and pass the cast object as an argument.
The line of my code (line 274)
		if(areCharctersAtSamePosition(&player, (Character *) itemPtr))
produces the following errors 
"
sccz80:"src/main.c" L:274 Error:Unknown member: _x
sccz80:"src/main.c" L:274 Error:Missing token, expecting ) got C
sccz80:"src/main.c" L:274 Warning:Too few arguments to call to function 'isCharacterAtLocation'
sccz80:"src/main.c" L:275 Error:Unknown member: _x
sccz80:"src/main.c" L:276 Error:Unexpected end of file
sccz80:"src/main.c" L:276 Error:Unexpected end of file
sccz80:"src/main.c" L:276 Warning:Converting pointer to integer without cast
sccz80:"src/main.c" L:274 Error:Unexpected end of file
sccz80:"src/main.c" L:274 Error:Unexpected end of file
sccz80:"src/main.c" L:274 Error:Unexpected end of file
sccz80:"src/main.c" L:274 Error:Unexpected end of file
sccz80:"src/main.c" L:274 Error:Unexpected end of file
"
The problem is avoided by using
"
		if(areCharctersAtSamePosition(&player, &(itemPtr->_character)))
"

The full function is:
```
void` handle_item(Item *itemPtr) 
{
	// Manage item
	if(itemPtr->_character._status == 1)
	{	
		if(areCharctersAtSamePosition(&player, (Character *) itemPtr))
		{
			itemPtr->_effect();
			powerUpReached((Character *) itemPtr);			
		}
		else
		{
			_blink_draw(itemPtr->_character._x, itemPtr->_character._y, itemPtr->_character._imagePtr, &(itemPtr->_blink));
		}		
	}
	else if (itemPtr->_coolDown == 0)
	{
		relocatePowerUp((Character *) itemPtr);

		_blink_draw(itemPtr->_character._x, itemPtr->_character._y, itemPtr->_character._imagePtr, &(itemPtr->_blink));
	}
	else
	{
		--itemPtr->_coolDown;
	}
}
```

suborb, 2018-01-03:
@Fabrizio-Caruso thanks for that, I can reproduce it now so I'll come up with a fix.

Fabrizio-Caruso, 2018-01-03:
Is the issue caused somehow by the macro?  The macro is OK for me.
By simmetry of the equality I can switch lhs and rhs in the macro and everything work.

suborb, 2018-01-03:
I think it may be, but I'm not sure how. It's worth fixing regardless, but if you're unblocked by switching the sides then that's good.

Fabrizio-Caruso, 2018-01-03:
@suborb, I am unblocked.

P.S.: Since I started re-writing my universal game in a more explicit object-oriented fashion I have opened issues on all the 8 bit cross compilers. Maybe I am pushing the cross compilers to their limits. It is a good thing, though because some issues will be solved and my game will be more memory efficient and more "generic".

suborb, 2018-01-07:
I merged this this evening, basically the cast wasn't happening at the right point in time.

closed 2018-01-07 by suborb