z88dk/z88dk #566: [VG5K] VG-5000 short of 1700 bytes vs BASIC

Fabrizio-Caruso, 2018-01-04:
When running VG-5000 binaries in an un-expanded VG-5000 (16k total RAM),
we only have about 12030 bytes of usable code.
On the other hand the BASIC interpreter provides 13758 bytes for the very same un-expanded machine.
There is a 1720 byte difference. How come? 
Is there a way to get a little more available memory?

Fabrizio-Caruso, 2018-01-05:
The following zpragma has no effect on the size:
"
// change program org
// #pragma output CRT_ORG_CODE = 24500   // an alternative to -zorg on the compile line

// move the stack to available memory
#pragma output REGISTER_SP = -1      // must be placed somewhere in the 16k ram

// halt program on exit, no need to play well with basic
#pragma output CRT_ON_EXIT = 0x10001

// eliminate the malloc heap
#pragma output CLIB_MALLOC_HEAP_SIZE = 0

// eliminate the stdio heap (no files can be opened)
#pragma output CLIB_STDIO_HEAP_SIZE = 0

// eliminate unused printf converters
#pragma printf = "%0u"    // placed used converters here, empty string if none
"

What can I try to reduce the size or gain extra available memory?

aralbrec, 2018-01-05:
The printf and CRT_ORG_CODE (or -zorg) will work but the other pragmas are newlib only.

Fabrizio-Caruso, 2018-01-05:
zorg=20000 provides 480 more bytes and seems to work fine!

Fabrizio-Caruso, 2018-01-05:
The game seems stable with zorg=20000. Probably it could be made start even lower.
This would explain the difference with BASIC.

Fabrizio-Caruso, 2018-01-05:
$49FD = 18941 should be a better default start location
Compare to 20480, we gain precious 1539 bytes.
This is empirical. I would prefer to use some docs, which I have not found, yet.

zx70, 2018-01-11:
I adjusted the default for zorg to 18950, you can still change this value with pragma or zorg

closed 2018-01-11 by zx70

joaopa00, 2018-03-05:
You can begin the program at 497f9H. Some commercial vg5000 games begin there if you need a lot of memory. Smaller commercial games begin at 5000H

zx70, 2018-03-05:
the choice of this lower location permits the autorun keeping as much memory as possible just after the basic.   the small adjustment was necessary for the single basic line calling the mc program to fit correctly.   It is still possible to choose custom locatios with the zorg parameter, but in that case the autorun feature is disablef and you have to provide your own loader or start the program manually.