z88dk/z88dk #567: Compilation error message with no file name detail on latest snapshot (2018/01/05)

Fabrizio-Caruso, 2018-01-06:
/cygdrive/c/z88dk/bin/zcc.exe +zx81 -O3 -vn -D__ZX81__ -DFULL_GAME -lndos -create-app -o  build/FULL_zx81_16k.prg src/zx81/zx81_graphics.c src/display_macros.c src/enemy.c src/invincible_enemy.c src/level.c src/character.c src/text.c src/missile.c src/strategy.c src/input_macros.c src/main.c
make: *** [Makefile:181: zx81_16k] Error 1

Remark:
The error is in file src/zx81/zx81_graphics.c
The Makefile is fine.

aralbrec, 2018-01-06:
Can you compile again with "-v" instead of "-vn" and capture the output?

Fabrizio-Caruso, 2018-01-12:
...
PROCESSING src/text.c
ucpp -iquote"." -DZ80 -DZX81 -D__ZX81__ -D__ZX81__ -DFULL_GAME  -DSCCZ80 -DSMALL_C -D__SCCZ80 -isystem"c:/z88dk//include"    "src/text.c" "C:\cygwin64\tmp\zcc64C18.i2"
zpragma -sccz80 < "C:\cygwin64\tmp\zcc64C18.i2" > "C:\cygwin64\tmp\zcc64C18.i"
sccz80  -ext=opt -mz80 -standard-escape-chars   "C:\cygwin64\tmp\zcc64C18.i"
copt "c:/z88dk//lib/z80rules.9" < "C:\cygwin64\tmp\zcc64C18.opt" > "C:\cygwin64\tmp\zcc64C18.op1"
copt "c:/z88dk//lib/z80rules.2" < "C:\cygwin64\tmp\zcc64C18.op1" > "C:\cygwin64\tmp\zcc64C18.opt"
copt "c:/z88dk//lib/z80rules.1" < "C:\cygwin64\tmp\zcc64C18.opt" > "C:\cygwin64\tmp\zcc64C18.op1"
copt "c:/z88dk//lib/z80rules.0" < "C:\cygwin64\tmp\zcc64C18.op1" > "C:\cygwin64\tmp\zcc64C18.asm"
type "C:\cygwin64\tmp\zcc64C18.asm" >> "C:\cygwin64\tmp\zcc64C18.tmp"
type "C:\cygwin64\tmp\zcc64C18.tmp" >> "C:\cygwin64\tmp\zcc64C18.asm"
z80asm --IXIY  -s  --cpu=z80 -I"c:/z88dk//lib" "C:\cygwin64\tmp\zcc64C18.asm"

PROCESSING src/missile.c
ucpp -iquote"." -DZ80 -DZX81 -D__ZX81__ -D__ZX81__ -DFULL_GAME  -DSCCZ80 -DSMALL_C -D__SCCZ80 -isystem"c:/z88dk//include"    "src/missile.c" "C:\cygwin64\tmp\zcc64C19.i2"
zpragma -sccz80 < "C:\cygwin64\tmp\zcc64C19.i2" > "C:\cygwin64\tmp\zcc64C19.i"
sccz80  -ext=opt -mz80 -standard-escape-chars   "C:\cygwin64\tmp\zcc64C19.i"
make: *** [Makefile:185: zx81_16k] Error 1

In this case the verbose mode tells me the problem is in src/missile.c
What kind of error?

Fabrizio-Caruso, 2018-01-12:
If I compile with CC65 I get the exact error:
src/missile.c(101): Error: Undefined symbol: `invincibleGhost'
src/missile.c(101): Error: Incompatible pointer types at 'Unknown'
src/missile.c(107): Error: Undefined symbol: `invincibleGhostHits'
src/missile.c(108): Error: Call to undefined function `decreaseGhostLevel'
src/missile.c(113): Error: Struct expected
src/missile.c(113): Error: Struct/union has no field named `_status'
src/missile.c(113): Error: Invalid lvalue in assignment
src/missile.c(114): Error: Struct expected
src/missile.c(114): Error: Struct/union has no field named `_x'
src/missile.c(114): Error: Struct expected
src/missile.c(114): Error: Struct/union has no field named `_y'
src/missile.c(114): Fatal: Too many errors
make: *** [Makefile:51: vic20_exp_16k] Error 1

suborb, 2018-01-12:
I’ll take a look at this tonight - is the code in your github? Alternatively if you can attach the preprocessed .i file (run zcc with the -E option) then that would be helpful.

Fabrizio-Caruso, 2018-01-12:
I have fixed the specific issue but the problem is general and appears on different errors.
So my GitHub page contains the fixed code that will compile with no errors.
I can try to reproduce it if you cannot investigate it otherwise.

suborb, 2018-01-12:
I reproduced something with the following:

```
void func()
{
        invincibleGhost->x = 1;
}
```

Which i guess from the cc65 error was what you were doing (unknown symbol being used as a struct) which gives the following errors now:

```
sccz80:missile.c L:6 Error:Unknown symbol: invincibleGhost
sccz80:missile.c L:6 Error:Non struct type can't take member
sccz80:missile.c L:6 Error:Must be lvalue
sccz80:missile.c L:6 Warning:Expected ';'
```

Previously only the first one was emitted and then it crashed which may have caused the OS not to show the first error.

closed 2018-01-12 by suborb