z88dk/z88dk #569: sccz80: doesn't like static declaration of a jagged array

aralbrec, 2018-01-06:
In the forum there is a [question about multi-dimensional arrays](https://www.z88dk.org/forum/viewtopic.php?id=10188).  It looks like sccz80 supports them now - is the implementation complete?

Anyway, I tried to statically create a jagged array to show as an alternative.

```
//unsigned char starField[12][64];

unsigned char _starField[12*64];

unsigned char *starField[12] = {
    &_starField[0],   // every 64 elements
    &_starField[64],
    &_starField[128],
    &_starField[192],
    &_starField[256],
    &_starField[320],
    &_starField[384],
    &_starField[448],
    &_starField[512],
    &_starField[576],
    &_starField[640],
    &_starField[704],
};

void main(void)
{
    unsigned char i,j;
    
    starField[2][20] = 10;
    
    for (i=0; i!=12; ++i)
        for (j=0; j!=64; ++j)
            starField[i][j] = 12;
}
```

sccz80 doesn't like the array of pointers:
`zcc +zx -vn -a -clib=new test.c`

but zsdcc accepts it:
`zcc +zx -vn -a -clib=sdcc_iy test.c`

suborb, 2018-01-06:
Yes, multi-dimensional arrays are officially supported now.

The issue above looks like an initialisation issue, I'll take a look at it.

sbebbers, 2018-01-06:
I guess that I need to upgrade to the latest nightly build?

suborb, 2018-01-06:
@sbebbers any nightly since about the end of October should support them, but latest is probably best!

suborb, 2018-01-07:
I've just merged #570 to resolve this issue

closed 2018-01-07 by suborb