z88dk/z88dk #574: Extra instructions produced when using #asm #endasm with __FASTCALL__

Ckirby101, 2018-01-14:
The compiler will output extra pop bc instructions depending on the compile order.

With the following code Func1 & Func 2 are correct but Func will add an extra pop bc which will cause a crash..

`
void __FASTCALL__ Func1()
{

}

void __FASTCALL__ Func2(int x)
{
	#asm
	ld a,l
	#endasm
}

void __FASTCALL__ Func3()
{

}
`

And this is the tmp output file.. 

`;void __FASTCALL__ Func1()
	C_LINE	107,"iso.c"
;{
	C_LINE	108,"iso.c"

; Function Func1 flags 0x00000208 __smallc __z88dk_fastcall 
; void Func1()
._Func1
	C_LINE	108,"iso.c"
;}
	C_LINE	110,"iso.c"
	ret

;void __FASTCALL__ Func2(int x)
	C_LINE	112,"iso.c"
;{
	C_LINE	113,"iso.c"

; Function Func2 flags 0x00000208 __smallc __z88dk_fastcall 
; void Func2(int x)
; parameter 'int x' at 2 size(2)
._Func2
	push	hl
	C_LINE	113,"iso.c"
;	#asm
	C_LINE	114,"iso.c"
	C_LINE	114,"iso.c"
	C_LINE	116,"iso.c"
	ld a,l
	C_LINE	117,"iso.c"
;}
	C_LINE	119,"iso.c"
	pop	bc
	ret

;void __FASTCALL__ Func3()
	C_LINE	122,"iso.c"
;{
	C_LINE	123,"iso.c"

; Function Func3 flags 0x00000208 __smallc __z88dk_fastcall 
; void Func3()
._Func3
	C_LINE	123,"iso.c"
;}
	C_LINE	125,"iso.c"
	pop	bc
	ret

`

Notice the extra pop bc in Func3

suborb, 2018-01-14:
Good spot, I think that's been there for a while. Hopefully the commit above should fix it.

Ckirby101, 2018-01-14:
cheers m8..

closed 2018-01-14 by suborb