z88dk/z88dk #579: [zsdcc][aquarius] Random not so random

Fabrizio-Caruso, 2018-01-27:
When I compile my game for the Aquarius target I am getting a non-random movement pattern of my elements whose movement choice is supposed to be either 0 or 1, which I select with
rand()&1 which should give 50% chance to 0 and 1.
It looks like it is almost always 0.

aralbrec, 2018-01-27:
The random number generator in the classic library is low quality.  I can see [looking at it](https://github.com/z88dk/z88dk/blob/master/libsrc/stdlib/rand.asm) that it will always generating alternating even and odd numbers so if you have an even number of calls to rand() you will almost always see 0 or 1 from rand()&1.

Maybe the classic can be updated to copy the one in newlib?  It's a marsaglia xor generator and is decent quality yet still fast:

[rand](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/stdlib/z80/asm_rand.asm) and [generator](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/stdlib/z80/random/asm_random_uniform_xor_32.asm)
[srand](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/stdlib/z80/asm_srand.asm)
[seed](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/stdlib/z80/__stdlib_seed.asm)

suborb, 2018-01-27:
I’m surprised that I haven’t lifted rand yet. It must be one of the few stdlib functions that isn’t a reference to newlib. 

I’ll take a look at it this evening.

suborb, 2018-01-27:
I’ve just switched classic over to the newlib rand implementation. Hopefully it will be a bit more random. It is a little larger than the old version though.

closed 2018-01-28 by suborb