z88dk/z88dk #580: [SVI][zsdcc] Loops for ever

Fabrizio-Caruso, 2018-01-29:
Compiling for the SVI target with zsdcc produces unstable code.

In particular I can see that a simple for loop just keeps looping forever:
```
	void slow_down(void)
	{
		short i;
		for(i=0;i<GAME_SLOW_DOWN;++i)
		{	
		}
	}
```

The very same code compiled with sccz80 (or cmoc or cc65) would just work.

aralbrec, 2018-01-30:
The generated code looks correct:

```
#define GAME_SLOW_DOWN 10000

void slow_down(void)
{
   short i;
   for(i=0;i<GAME_SLOW_DOWN;++i)
   {  
   }
}
```

zcc +zx -vn -a -clib=sdcc_iy -SO3 --max-allocs-per-node200000 test.c

```
_slow_down:
	ld	bc,0x2710
l_slow_down_00104:
	ld	e, c
	ld	d, b
	dec	de
	ld	c, e
	ld	a,d
	ld	b,d
	or	a,e
	jr	NZ,l_slow_down_00104
	ret
```

The loop counts from `GAME_SLOW_DOWN` down to 0.

How big is `GAME_SLOW_DOWN` ?  It can't exceed 32767 because then it becomes a negative number.

Fabrizio-Caruso, 2018-01-30:
GAME_SLOW_DOWN is much smaller than 2^15-1.
Nevertheless, the game gets stuck there (I print before and after some debug string).
I don't know how to reproduce it in an isolated way.

suborb, 2018-01-30:
Given that this can't be reproduced in an isolated way, I'm guessing this loop is part of a larger bit of code?

If so, can you run with -cc -a and provide the output .asm file?

Fabrizio-Caruso, 2018-01-31:
I have now re-tested this issue on a very recent version of Z88DK.
The issue is different now. It seems related to psg code. 
Consistently I have:
sccz80 with psg -> OK
zsdcc without psg -> OK
zsdcc with psg -> KO

By psg I mean functions in #include <psg.h> such as set_psg.

I will try to reproduce and possibly isolate the issue. I will open a different issue.

This issue is no longer valid. Sorry.

closed 2018-01-31 by Fabrizio-Caruso